PX4/PX4-Autopilot #18928: sitl of gazebo waitting in preparing your world

mengchaoheng, 2021-12-31:
after v1.12.0 the time of preparing is very long.what have you change?

Jaeyoung-Lim, 2022-01-02:
@mengchaoheng Please explain more on what setup you are running on (OS, gazebo version etc) and the log of the issue you are experiencing

If you want more outputs, you can add a `VERBOSE_SIM=1` in front of your make commands to get more informaiton

mengchaoheng, 2022-01-02:
ubuntu 18.04，gazebo9.19

I have downloaded the model offline to the .gazebo folder. My latest test found that no matter what version of px4 (v1.11 or v1.12), when doing "make px4_sitl gazebo", it stays at the "preparing your world" interface. But this situation did not happen a few months ago. The only possibility is that I started to develop ros in the last month and installed other dependent environments, which led to this "preparing your world"

===========================
mch@mch-virtual-machine:~/DF/PX4-Autopilot$ VERBOSE_SIM=1 make px4_sitl gazebo
-- PX4 version: v1.12.0
-- PX4 config file: /home/mch/DF/PX4-Autopilot/boards/px4/sitl/default.cmake
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is GNU 7.5.0
-- The C compiler identification is GNU 7.5.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/cc
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.9", minimum required is "3") 
-- build type is RelWithDebInfo
-- PX4 ECL: Very lightweight Estimation & Control Library v1.9.0-rc1-569-g71fc1b8
-- ROMFS: ROMFS/px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /home/mch/DF/PX4-Autopilot/build/px4_sitl_default
[0/817] git submodule src/drivers/gps/devices
[1/817] git submodule src/lib/ecl
[2/817] git submodule mavlink/include/mavlink/v2.0
[4/817] git submodule Tools/sitl_gazebo
[15/817] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is GNU 7.5.0
-- The CXX compiler identification is GNU 7.5.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Looking for pthread.h
-- Looking for pthread.h - found
[... 293 lines cut ...]
INFO  [ecl/EKF] 4140000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 4284000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 4340000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 4344000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] 4344000: EKF aligned, (baro hgt, IMU buf: 12, OBS buf: 9)
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [ecl/EKF] reset position to GPS
INFO  [ecl/EKF] reset velocity to GPS
INFO  [ecl/EKF] starting GPS fusion
INFO  [tone_alarm] home set
INFO  [tone_alarm] notify negative

mengchaoheng, 2022-01-02:
[image]

[image]

mengchaoheng, 2022-01-02:
> 
can you help me ? thanks

Jaeyoung-Lim, 2022-01-02:
> My latest test found that no matter what version of px4 (v1.11 or v1.12),

This means that this is not related to the firmware.

Please try to remove `.gazebo` and try again if this solves the problem

mengchaoheng, 2022-01-14:
> > My latest test found that no matter what version of px4 (v1.11 or v1.12),
> 
> This means that this is not related to the firmware.
> 
> Please try to remove `.gazebo` and try again if this solves the problem
> 
> Also I can see that you have some problems in your dependency chain. e.g.
> 
> ```
> -- Building OpticalFlow with OpenCV
> Traceback (most recent call last):
>   File "/usr/bin/rosversion", line 6, in <module>
>     from pkg_resources import load_entry_point
>   File "/home/mch/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3242, in <module>
>     @_call_aside
>   File "/home/mch/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3226, in _call_aside
>     f(*args, **kwargs)
>   File "/home/mch/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 3255, in _initialize_master_working_set
>     working_set = WorkingSet._build_master()
>   File "/home/mch/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 568, in _build_master
>     ws.require(__requires__)
>   File "/home/mch/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 886, in require
>     needed = self.resolve(parse_requirements(requirements))
>   File "/home/mch/.local/lib/python3.6/site-packages/pkg_resources/__init__.py", line 772, in resolve
>     raise DistributionNotFound(req, requirers)
> pkg_resources.DistributionNotFound: The 'rospkg==1.3.0' distribution was not found and is required by the application
> ```
> 
> Also try building in a clean environment by running
> 
> ```
> make distclean
> ```

I've tried both, but none of them work. I had to redeploy the development environment on the new system.