PX4/PX4-Autopilot #18940: Mixer Development in PX4 and actuator control for Mavros

Jawad-RoboLearn, 2022-01-03:
Hello, 
I am developing the PX4-Mavros based software for my unconventional quadrotor. For this purpose, I really need to work at Forces/Body torques level. I am planning to use mavros>>actuator_setpoint topics. 

I went through:
1. Mixers.mix files
2. <geometry>.toml files for different multicopter geometries.
3.  The MulticopterMixer.cpp files and I understood what's going on. 

Now the concerns are:
1. I have to pass the geometry dependent variables in the control allocation mapping matrix ---> The matrix relates F/tau and (roll/pitch/yaw/thrust). In MulticopterMixer.cpp file its feasible to transform the matrix. 
`const MultirotorMixer::Rotor _config_quad_x[] = {
	{ -0.707107,  0.707107,  1.000000,  1.000000 },
	{  0.707107, -0.707107,  1.000000,  1.000000 },
	{  0.707107,  0.707107, -1.000000,  1.000000 },
	{ -0.707107, -0.707107, -1.000000,  1.000000 },`

2. I am using the quad_x mixer and plan to include the changes in "MulticopterMixer.cpp". In quad_x.toml file the rows of control_allocation matrix are provided and they are fixed. Actually the "position" variables below for each rotor are not fixed in my case. Do I need to parse some variables in .toml file, or only changes in MulticopterMixer.cpp are ok?

`# Generic Quadcopter in X configuration

[info]
key = "4x"
description = "Generic Quadcopter in X configuration"

[rotor_default]
direction = "CW"
axis      = [0.0, 0.0, -1.0]
Ct        = 1.0
Cm        = 0.05

[[rotors]]
name      = "front_right"
position  = [0.707107, 0.707107, 0.0]
direction = "CCW"

[[rotors]]
name      = "rear_left"
position  = [-0.707107, -0.707107, 0.0]
direction = "CCW"

[[rotors]]
name     = "front_left"
position = [0.707107, -0.707107, 0.0]

[[rotors]]
name     = "rear_right"
position = [-0.707107, 0.707107, 0.0]`

3. Please suggest changes in firmware required when we are writing the custom mixers file. Like updating CMakeLists, or any other folder in PX4-Autopilot. Does changing this mixer file applicable to codes written in OFFBOARD mode in Mavros.

Thanks.

Jawad-RoboLearn, 2022-01-03:
@Jaeyoung-Lim @dagar @bresch @mhkabir . I am tagging you guys because I have learned a lot from your repos and comments in different issues.

Jaeyoung-Lim, 2022-01-10:
@Jawad-RoboLearn Without the custom mixer, are you able to control a normal quadrotor with actuator commands?

Normally it is not recommended that you send high bandwidth actuator commands over mavlink on a real system with the time delay and jitter introduced with serializing / deserializing commands in and out of mavlink.

Jawad-RoboLearn, 2022-01-10:
@Jaeyoung-Lim I am not using the direct actuator commands here. I was trying to add the mixer file of my custom drone and use /mavros/actuator_controls (F/tau) computed by my control. However it didn't work currently. 

Yes, I have understood the "Direct Actuator Control" git by one of the member here in SITL and also read the discussion. Thanks. 

I have one question from you regarding Mavros Offboard control with thrust and attitude outputs on /mavros/attitude_topic (sorry I forgot the name) 

• I am trying to implement controller in real time. Are there some precautions I need to take. Like manually adding my quadcopter (QAV250) geometry.toml file or any other thing. 
• If I use the attitude and thrust setpoint to my quad, does it still using some internal PX4 controller in Offboard mode?

Jaeyoung-Lim, 2022-01-10:
> @Jaeyoung-Lim I am not using the direct actuator control here. I was trying to add the mixer file of my custom drone and use /mavros/actuator_controls (F/tau) computed by my control. However it didn't work currently.

I am referring to this specific topic. It is not recommended that you use this topic for high bandwidth control. I am not saying it is not possible, but it would be hard to get competitive performance compared to implementing the controller directly inside the FMU

> • I am trying to implement controller in real time. Are there some precautions I need to take. Like manually adding my quadcopter (QAV250) geometry.toml file or any other thing.

You have to make sure everything is working :smile: Not sure what you are asking here

> • If I use the attitude and thrust setpoint to my quad, does it still using some internal PX4 controller in Offboard mode?

Yes, the attitude loop depending on what setpoint you are using is being used that is running inside PX4

closed 2022-01-10 by Jawad-RoboLearn