PX4/PX4-Autopilot #18941: Multi-Vehicle Gazebo Sim Vehicle 2 Not Arming with ROS2

adk4davsys, 2022-01-03:
**Describe the bug**
During multi-vehicle gazebo simulation with two vehicles, only the first vehicle accepts the command. ROS2 is used to arm the vehicle for takeoff but the second vehicle ignores all the commands.

**To Reproduce**
Steps to reproduce the behavior:
1. Launch PX4 multi-vehicle SITL using gazebo ("-n 2" flag is used to instantiate two vehicles)
2. Run ROS2 nodes with micrortps
3. Command the vehicles to arm after changing the mode to offboard using QGC
4. Observe that only the first vehicle accepts the incoming command and arms while the second vehicle ignores the command

**Expected behavior**
The second vehicle should also arm when the command is sent from ROS2 node for takeoff.

**Drone (please complete the following information):**
- gazebo SITL using default iris model
- using PX4 version 1.12.3

**Additional context**
I also tried sending landing command using a ROS2 node and it gets ignored by the vehicle as well. Currently I can manually send the command from QGC for arming and landing but it would be nice to use ROS2 commands.

Any pointers to possible solution are much appreciated. Thank you in advance!

closed 2022-03-07 by adk4davsys

aomezzine, 2022-04-05:
Hi,
I have exactly the same problem described above. Could you please give me more details about the solution ? where can we specify that value ? 

Thank you

adk4davsys, 2022-04-05:
> Hi, I have exactly the same problem described above. Could you please give me more details about the solution ? where can we specify that value ?
> 
> Thank you

Hi,

If I remember right, all I had to do was set `target_system` according to the number of simulated vehicles. `target_system=1` for the first vehicle, `target_system=2` for the second, and so on so forth. I am using [this](https://github.com/PX4/px4_msgs/blob/master/msg/VehicleCommand.msg) topic with the command `x4_msgs::msg::VehicleCommand::VEHICLE_CMD_COMPONENT_ARM_DISARM`.

aomezzine, 2022-04-05:
Hi,
It is ok :)
Thank you