PX4/PX4-Autopilot #18954: macos (M1) make sitl gazebo get error

liganggg, 2022-01-05:
hi everyone 

I get error when I build make px4_sit gazebo on Mac OS  

i use MacBook Air (M1, 2020)
gezebo version is Gazebo multi-robot simulator, version 9.19.0

It can be compiled normally when  make px4_fmu-v5_default or make px4_sitl jmavsim
 
but when i make px4_sitl gazebo 
i get error：
 
`fatal error: too many errors emitted, stopping now [-ferror-limit=]
20 errors generated.
ninja: build stopped: subcommand failed.
[1/5] Generating ../../logs
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build /Users/l/px4/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/l/px4/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.22.1/bin/cmake --build /Users/l/px4/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 0
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1`

I have tried some methods found on the Internet but it don't work

I need some help Anyone get this error and recommendation? thanks!

Jaeyoung-Lim, 2022-01-05:
@liganggg Probably you might get some hints from https://github.com/PX4/PX4-Autopilot/pull/18617

liganggg, 2022-01-06:
> @liganggg Probably you might get some hints from #18617

thanks for you advice ,i have tryed reinstall homebrew-core and brew tap osrf/simulation 
but still get the same error，i guess it maybe related with the macos monterey 12.1. or gazebo version?
Can someone help try the same environment,and give some suggestion thanks ~

liganggg, 2022-01-07:
```
adsfjkaf
adsfgd
```

liganggg, 2022-01-11:
these days i have tryed to use anaconda instead python whitch mac raw version，bug the same error still exist.

liganggg, 2022-01-13:
thanks everyone  
 i had solve this problem by restore all content and settings,and then install step by step flow 'px4 develop guide' 
The previous error may be some step has  mistake operation

closed 2022-01-13 by liganggg