PX4/PX4-Autopilot #18957: VEHICLE_CMD_DO_SET_PARAMETER not implemented

benjinne, 2022-01-05:
**Describe problem solved by the proposed feature**
When attempting to use RTPS to set parmeters, the VEHICLE_CMD_DO_SET_PARAMETER command is not implemented.
https://github.com/PX4/PX4-Autopilot/blob/0a82025fafc7275caca731a9672f74fa3b55b2b8/msg/vehicle_command.msg#L33

**Describe your preferred solution**
It appears the mavlink driver directly sets parameters here:
https://github.com/PX4/PX4-Autopilot/blob/0a82025fafc7275caca731a9672f74fa3b55b2b8/src/modules/mavlink/mavlink_parameters.cpp#L131
This is unlike other mavlink commands that publish to the vehicle_command topic and are handled by the commander. I'm just pointing this out, not that I suggest it be changed. I do request the commander handle VEHICLE_CMD_DO_SET_PARAMETER so that I can set parameters using offboard RTPS.

**Additional context**
Attempting to set parameters using offboard with RTPS doesn't appear possible, and it seems like MAVlink is the only offboard solution to setting parameters.

dagar, 2022-01-05:
This doesn't help you immediately, but I plan to update the entire parameter system to have an optional backend that's much more like a ROS2 compatible parameter server.

drasgo, 2022-11-09:
Hi, any plan in supporting VEHICLE_CMD_DO_SET_PARAMETER for rtps, or if the decision is to go directly for the full update, any idea on how long it will take?

dagar, 2022-11-09:
> Hi, any plan in supporting VEHICLE_CMD_DO_SET_PARAMETER for rtps, or if the decision is to go directly for the full update, any idea on how long it will take?

I've been experimenting with making PX4 parameters fully accessible to ROS2, but I'm not sure when I'll be able to finish it.

Much more immediately we could probably implement `MAV_CMD_DO_SET_PARAMETER` if you think it's important. I find that interface quite awkward because you need to separately determine the parameter number and then the parameter value type is a mess, but I suppose it's better than nothing.

hamishwillee, 2024-04-10:
@dagar @benjinne There is a proposal to deprecate `MAV_CMD_DO_SET_PARAMETER` in https://github.com/mavlink/mavlink/pull/2099 with the intent to eventually remove - since there are no implementations and it has been around forever. Any objections?

FWIW I can definitely see some cases where `MAV_CMD_DO_SET_PARAMETER` is useful, and now that you can use component metadata protocol in MAVLink to get all the parameter info - names, data types etc, it isn't a stretch to use this safely and relatively easily with PX4.

Generally though, we prefer providing specific interfaces rather than general purpose setters for "internal" things such as parameters.

closed 2025-10-21 by hamishwillee