PX4/PX4-Autopilot #18961: offboard position or attitude setpoints for fixedwing

airsimdevdd, 2022-01-06:
does the fixedwing supports position or attitude setpoints any example code snippet

Jaeyoung-Lim, 2022-01-06:
@airsimdevdd You xan what setpoints are supported Brisket Southern BBQ & Bar
044 241 20 20
https://maps.app.goo.g

airsimdevdd, 2022-01-06:
sir any example code for fixedwing offboard @Jaeyoung-Lim

Jaeyoung-Lim, 2022-01-06:
@airsimdevdd https://mavsdk.mavlink.io/main/en/cpp/examples/offboard_velocity.html

closed 2022-01-06 by Jaeyoung-Lim

airsimdevdd, 2022-01-07:
offboard_position_ned.py is not working for fixed wing ,its is showing no rc, no offboard failsafe activated

Jaeyoung-Lim, 2022-01-07:
@airsimdevdd Your problem is not related to this issue. Please provide a detailed explanation on how you are testing so that someone can reporduce your problems. Otherwise it is hard to find out what your problem is.

Are you testing in SITL Gazebo? I think your problem is not related to offboard, but you don't have RC Failsafe disabled for offboard. Have you checked if the problem goes away if you have a virtual joystick in QGC?

airsimdevdd, 2022-01-07:
yes i was testing in sitl gazebo for and rc 
failsafe disabled

Jaeyoung-Lim, 2022-01-07:
@airsimdevdd Is it fixed now?

airsimdevdd, 2022-01-08:
yes it is fixed positionnedyaw works fine but velocitynedyaw seems not working properly

gurselturkeri, 2024-03-15:
> @airsimdevdd
> 
>     * The PX4 documentation also explains with a code snippet: http://docs.px4.io/master/en/ros/mavros_offboard.html
> 
>     * mavsdk cpp has some examples: https://mavsdk.mavlink.io/main/en/cpp/examples/offboard_velocity.html
> 
>     * mavsdk python also has a bunch of examples such as https://github.com/mavlink/MAVSDK-Python/blob/main/examples/offboard_position_ned.py

Is there any offboard mavros example for fixed wing SITL gazebo?