PX4/PX4-Autopilot #18965: Violation of acceleration constraints (anomalous states reported in the flight log)

squizz617, 2022-01-06:
**Describe the bug**
Similar to issue #18962, a certain sequence of manual control commands can make a PX4 drone violate acceleration limits. In this particular test case, `MPC_ACC_HOR_MAX` constraint is violated, along with UP and DOWN constraints.

**To Reproduce**
Steps to reproduce the behavior:
1. [terminal 1] make px4_sitl_rtps gazebo -j4
2. [terminal 2] Arm and send the following manual_control messages over MAVLink every 0.1 seconds: [1641443932.2358694-controls.txt](https://github.com/PX4/PX4-Autopilot/files/7823682/1641443932.2358694-controls.txt)

**Expected behavior**
The acceleration of the drone should stay within the boundaries set by the parameters.

**Log Files and Screenshots**
- [Flight Review](https://review.px4.io/plot_app?log=b700c57d-b8ff-4a30-969a-451b7fed3cb1)
- Sudden spikes in raw acceleration are observed at time 0:18, where acc_x = 12.23, acc_y = -13.82, and acc_z = -22.46. 

**Additional context**
Those manual control commands are somewhat "radical" as they are generated by a fuzzer, but are in the boundaries expressible by a joystick. I tried, but couldn't understand how exactly these commands cause the anomalous state of the drone. Would you help shed light on this? Thank you.

bresch, 2022-01-12:
To me, it looks like the drone is just hitting the ground, no?
[image]

squizz617, 2022-01-12:
Ah, that makes a lot of sense. I didn't check the Z position. Thanks!

closed 2022-01-12 by squizz617