PX4/PX4-Autopilot #18968: mRo OEM H7 GPS doesn’t work

mrpollo, 2022-01-06:
Using an OEM H7 you can't get a GPS to work, using the GPS port or CAN, neither show up with `gps status`

```
NuttShell (NSH) NuttX-10.1.0
nsh> gps status
INFO
(gps] Main GPS
INFO
[gps] protocol: UBX
INFO
(gps] status: NOT OK, port: /dev/ttys2, baudrate: 0
INFO
[gps] sat info: disabled
INFO
gps] rate reading:
0 B/s
nsh>
```

ryanjAA, 2022-01-06:
Will try next week (when back in the office) are you on master?

mrpollo, 2022-01-10:
Yep master and stable are the same @ryanjAA did you get a chance to test this?

ryanjAA, 2022-01-10:
Ok - the H7 and F7 Oem I have don't have the boot loaders on them (just tried) so I'll find a segger tomorrow and get it burnt on. I have to track down the little 6 pin pogo connector (in a new office but I do have one). Standby.

dagar, 2022-01-10:
> Ok - the H7 and F7 Oem I have don't have the boot loaders on them (just tried) so I'll find a segger tomorrow and get it burnt on. I have to track down the little 6 pin pogo connector (in a new office but I do have one). Standby.

You don't have any bootloader loaded or you don't have the PX4 bootloader?

ryanjAA, 2022-01-11:
There are no boot loaders on them - it's because awhile back when you and I were working on getting them functional, the boot loaders were put on and off, etc before migrating to the non OEM variants for use which is predominantly what we use for Control Zeros.

taileron, 2022-01-11:
with the by mro preinstalled AP bootloader gps works flawlessly on all white plug uart ports available on the stick adapter  ...

ryanjAA, 2022-01-12:
@mrpollo I put the PX4 bootloader on and on Master I get sats. On stable when I go into the console and hit return it sets me into a reboot. To @dagar on that one...

I will try and just see if it gets sats now on stable but it does work (on master). didn't try the AP boot loader but it sounds like that is working, I can try if need be.

benjinne, 2022-01-15:
I'm also not getting sats on my pixracer pro

benjinne, 2022-01-16:
I found the commit that fixes it: https://github.com/PX4/PX4-Autopilot/commit/1ee3484827a05188fd415f308adcf7abd121b8f5
I found it by trying older and newer commits until I found the commit where it goes from not working to working

dirksavage88, 2022-01-24:
It seems the betaflight crowd has had issues with the ublox m9n as well: https://github.com/betaflight/betaflight/pull/10921

dagar, 2022-01-24:
To be clear is this now working on master for everyone? If not please specify the specific hardware, PX4 commit, and wiring (ctrl zero specifics).

benjinne, 2022-01-24:
This works on master, but is broken on latest release. I found the commit https://github.com/PX4/PX4-Autopilot/commit/1ee3484827a05188fd415f308adcf7abd121b8f5 fixes it. I'm running the latest release so I just cherry-picked that commit in and I've been flying with it no problem

benjinne, 2022-03-18:
> This works on master, but is broken on latest release. I found the commit [1ee3484](https://github.com/PX4/PX4-Autopilot/commit/1ee3484827a05188fd415f308adcf7abd121b8f5) fixes it. I'm running the latest release so I just cherry-picked that commit in and I've been flying with it no problem. I'm running a pixracer pro

I just found that works with the pixracer pro, doesn't work with the ctrl-zero-h7-oem. The latest master branch works, but as expected the latest release v1.12.3 does not

benjinne, 2022-03-20:
This commit https://github.com/PX4/PX4-Autopilot/commit/3f17acdcc905c75aef849156107d19e17e097434 fixes the ctrl-zero-h7-oem along with previous commit I found that fixed it for the pixracerpro. This commit can't be cherry-picked into v1.12.3 due to file structure changes, however I manually made the changes into my v1.12.3 branch and it works now.

ryanjAA, 2022-03-20:
Good to know for the time being. You can create a patch and post it here for people.

ryanjAA, 2022-03-20:
@mrpollo i think this is what you were mentioning before.

JacobCrabill, 2022-03-21:
Possibly related issue - has anyone got the TELEM1 port of the prototype breakout board (USART2) working?  As far as I can tell, I can send data out from PX4 on that port, but Rx data is not being captured.  I can't find anything obviously wrong in the pinout of the prototyping board, but without schematics, I can't say for certain.

ryanjAA, 2022-03-21:
Which breakout board? The square or rectangle one? I have them both so can give it a try. Are you on master or release?

JacobCrabill, 2022-03-21:
I'm making some changes off of master; I haven't updated in a week or so, but still quite recent.

I'm using their prototype carrier board (https://store.mrobotics.io/product-p/m10060b.htm), the larger, more square-ish board with just USB, GPS1, TELEM1, CAN1, and CAN2 connectors, plus all connections broken out to holes for header pins.  I've already found another oddity with the TELEM1 port, which is that the RTS pin of TELEM1 seems to be not connected to anything.

ryanjAA, 2022-03-21:
I'll take a look - strange about the RTS. Does rts have continuity on the breakout board itself (from the port to header) or do you think it is on the actual autopilot? I have the autopilot schematics so let me find them and look (although I think I have the f7 schematics but should be the same iirc).

JacobCrabill, 2022-03-21:
> I'll take a look - strange about the RTS. Does rts have continuity on the breakout board itself (from the port to header) or do you think it is on the actual autopilot? I have the autopilot [...]

Sorry, I meant to say it's a continuity issue on the breakout board itself.  I haven't go so far as to actually toggle pins on the MCU and verify the actual pinout of the H7 OEM board between the MCU and the 2 connectors.

Schematics would be extremely helpful, if you're able to share them!

ryanjAA, 2022-03-21:
Gotcha. Ok. Not on the autopilot then. I’ll take a look on what I have. 

Can’t share the schematics unfortunately as they sent them over to me for an issue with the FC a long while back but happy to look through anything as needed. 

Let me see on master what pops up. If I can’t find the correct board I’ll grab one tomorrow at the office and check but always good to have two people try the same thing to rule out something specific to the hardware too. Standby - let’s see.

JacobCrabill, 2022-03-21:
> Can’t share the schematics unfortunately as they sent them over to me for an issue with the FC a long while back but happy to look through anything as needed.

Ok, I kinda thought so.  If you could just confirm that the USART2 Rx/Tx pins are correctly setup in PX4's board.h, that would be really helpful.  I'm struggling to determine whether there's an obscure issue in PX4 preventing Rx from working, or whether it's a hardware issue on the carrier board I'm not seeing, or something else.

ryanjAA, 2022-03-23:
Telem1 port is working fine on that breakout board here. Put mater on from right now and a generic sik radio and connected wirelessly. Changed a few params and rebooted, all working.

ryanjAA, 2022-03-26:
@JacobCrabill hardware is specific to your batch. See this note. 

https://px4.slack.com/archives/C0W2KUFFT/p1648319405997709?thread_ts=1648302068.722429&channel=C0W2KUFFT&message_ts=1648319405.997709