PX4/PX4-Autopilot #18969: micrortps_client and micrortps_agent UDP connection not working

evbernardes, 2022-01-07:
**Describe the bug**
px4_ros_com successfully compiled inside my ROS2 (galactic) workspace, and activated on the PX4 side. 
I manage to make a connection between by UART over USB, but UDP connection never works and nothing is sent.
Tried both with access mode and station mode for the WiFi.

**To Reproduce**
This works:
1. Connect computer to pixracer over USB
2. On PX4 shell: `micrortps_client start -t UART -d /dev/ttyACM0 -b 460800`
3. On my computer terminal: `micrortps_agent -d /dev/ttyACM0 -b 460800`
4. Opening rqt I can see the messages being sent

This DOES NOT work:
1. Connect both pixracer and computer with fixed ip adresses to the same network over wifi.
2. On PX4 shell: `micrortps_client start -t UDP -i <ip.adress.of.COMPUTER>`
3. On my computer terminal: `micrortps_agent start -t UDP -i <ip.adress.of.PIXRACER>`
4. Opening rqt, nothing is exchanged

**Drone (please complete the following information):**
- mRobotics Pixracer board with up-to-date PX4 Firmware
- ESP8266 wifi module 
- Ubuntu 20.04
- ROS2 Galactic

**Additional context**
As another unexpected behaviour, even though I correctly set the ip adress of the drone, I get the message:
`1641562296.353968 [0] micrortps_: using network interface eno1 (udp/ANOTHER.IP.ADRESS) selected arbitrarily from: eno1, wlp111s0`, while eno1 is the ethernet interface my computer uses to connect to the internet while wlp111s0 is the wifi interface it should be using. 
Disconnecting the ethernet and relaunching micrortps_agent removes this message, but still doesn't work.

benjinne, 2022-01-15:
I'm not sure how to fix your issue, but is mavlink working with the wifi module? Also, the message about an interface being selected arbitrarily is a standard message when using cyclone DDS. You can try switching to fast DDS and see if that helps.

evbernardes, 2022-01-16:
> I'm not sure how to fix your issue, but is mavlink working with the wifi module? Also, the message about an interface being selected arbitrarily is a standard message when using cyclone DDS. You can try switching to fast DDS and see if that helps.

Thanks for your reply! Sadly since I didn't have any replies until now and I have a time constraint, I had to give up and install ROS1 Noetic with Mavros.