PX4/PX4-Autopilot #18972: OpenJDK-14 No Longer Exist on any Ubuntu Repos

jsantos18, 2022-01-07:
# Issue

Building PX4-Autopilot using the https://github.com/PX4/PX4-Autopilot/blob/master/Tools/setup/ubuntu.sh script it will fail due to Canonical Taking OpenJDK-14 off the repos. Referenced here https://packages.ubuntu.com/search?keywords=openjdk&searchon=names&suite=all&section=all

# Proposed Solution

Either downloading the openjdk-14 from source or Upgrading to a LTS? if there even is one of openjdk

julianoes, 2022-01-08:
Which Ubuntu version are you referring to?

jsantos18, 2022-01-10:
I was using both 21.04 and 21.10

julianoes, 2022-01-10:
Aha, we don't actively support newer than 20.04. But you could make a PR to the setup.sh script that works for your newer versions, that would be great.

mrpollo, 2022-01-25:
@jsantos18 try JDK 17 for now, I'm working on a revamped version of ubuntu.sh