PX4/PX4-Autopilot #18982: Detect if drone is in air on offboard mode

DevJav, 2022-01-10:
I have a self made state machine for controlling missions. One of the states is "armed", and I'm trying to find a way to detect that the drone did the take off after sending a waypoint to jump to a new state. I tried by checking the z-axis but it may not be the most efficient way to check it as you may not always take off starting from 0.

antonio-sc66, 2022-01-10:
Hi @DevJav, I think that checking the _landed_state_ field of the _ExtendedState_ message should be enough. You need to make sure it's value is _LANDED_STATE_IN_AIR_

http://docs.ros.org/en/noetic/api/mavros_msgs/html/msg/ExtendedState.html

DevJav, 2022-01-11:
Hi @antonio-sc66. I didn't know about that message so thank you!

However I've been making some experiments with the simulator and it seems like if you add a waypoint to make it take off, it changes to LANDED_STATE_IN_AIR immediately, which it's not true as it takes some time before it actually stops touching the ground.

benjinne, 2022-01-15:
Maybe the `takeoff_status` topic is what you're looking for. https://github.com/PX4/PX4-Autopilot/blob/master/msg/takeoff_status.msg It gets updated in the mc_pos_control module

DevJav, 2022-01-17:
> 

I'm actually using ROS1 with mavros and this topic dosn't seem to be published :(

ytaskiran, 2022-02-04:
I do not know your exact use case and what is your aim but here are my suggestions:

- Maybe you can try LANDED_STATE_TAKEOFF message instead of IN_AIR.
- You can also look at WaypointReached message of mavros if you do the takeoff with waypoints so that you can understand the current state.