PX4/PX4-Autopilot #18983: RTL instability when VEHICLE_CMD_DO_SET_HOME is used with ROS2

adk4davsys, 2022-01-10:
**Describe the bug**
When triggering RTL during HITL with VEHICLE_CMD_DO_SET_HOME results in instability. I included screenshots but it seems to me that the vehicle still points to the origin while flying towards the new home location.

**To Reproduce**
Steps to reproduce the behavior:
1. Using the offboard mode, set new home position (offset latitude by 100 meters) using VEHICLE_CMD_DO_SET_HOME (only works after ARMING)
2. Takeoff and fly to an arbitrary location
3. Trigger RTL
4. Observe vehicle flies toward the new HOME location while twitching
5. Landing is very hazardous as the vehicle keeps bouncing
6. After landing, vehicle does not DISARM automatically

**Expected behavior**
Vehicle should fly towards HOME in a stable manner.

**Log Files and Screenshots**
[image]
[image]

**Drone (please complete the following information):**
- Using V1.12.3, not master branch
- When using RTL without setting new HOME, it flies fine.

dagar, 2022-01-11:
Is VEHICLE_CMD_DO_SET_HOME being sent multiple times? Can you share the full log? https://logs.px4.io

adk4davsys, 2022-01-11:
@dagar Thank you for your reply. Attached is the log from above flight, please let me know if you need anything else.

[log](https://review.px4.io/plot_app?log=a77dd16c-01bc-41be-a272-ec70dc880002)

adk4davsys, 2022-01-13:
@dagar Seems like when using VEHICLE_CMD_DO_SET_HOME affects Position Mode as well. I tried setting home using `current location=1` once and switching to Position Mode does the same thing (setpoint is set to local origin). Below is the screenshot of my log:

[image]

Can someone else replicate this behavior by using VEHICLE_CMD_DO_SET_HOME? I would like to use this feature but seems like it is very hazardous at the moment.

adk4davsys, 2022-01-13:
@dagar Turns out `trajectory_setpoint` topic from companion computer still gets subscribed and applied during RTL. Not publishing to this topic when  RTL is triggered the problem does not persist. I will open a new issue and reference this issue.

closed 2022-01-13 by adk4davsys