PX4/PX4-Autopilot #18999: Unable to '"make px4_sitl gazebo"

Jeswin17, 2022-01-12:
I followed the instructions given here https://docs.px4.io/master/en/dev_setup/dev_env_linux_ubuntu.html to setup Gazebo and PX4.
But when it try to run gazebo, it throws the following error.

**Env:**
Ubuntu 20.04.3 LTS
Gazebo 9 (  d/led when I ran the https://github.com/PX4/PX4-Autopilot/blob/master/Tools/setup/ubuntu.sh script as mentioned in the docs.

> -- Found GStreamer: adding gst_video_stream_widget
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
Boost_INCLUDE_DIR
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
  -- Configuring incomplete, errors occurred!
  See also "/home/VMuser/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
  See also "/home/VMuser/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
  [247/846] ROMFS: copying, generating airframes
  FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
  cd /home/VMuser/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/VMuser/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/VMuser/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
  [249/846] Generating parameters.xml
  ninja: build stopped: subcommand failed.
  make: *** [Makefile:235: px4_sitl] Error 1

**The output of the CMakeError.log file is as follows:**

> 
> Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
> Change Dir: /home/VMuser/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/bin/ninja cmTC_f4056 && [1/2] Building C object CMakeFiles/cmTC_f4056.dir/src.c.o
> [2/2] Linking C executable cmTC_f4056
> FAILED: cmTC_f4056 
> : && /bin/cc -DCMAKE_HAVE_LIBC_PTHREAD   CMakeFiles/cmTC_f4056.dir/src.c.o  -o cmTC_f4056   && :
> /bin/ld: CMakeFiles/cmTC_f4056.dir/src.c.o: in function `main':
> src.c:(.text+0x46): undefined reference to `pthread_create'
> /bin/ld: src.c:(.text+0x52): undefined reference to `pthread_detach'
> /bin/ld: src.c:(.text+0x63): undefined reference to `pthread_join'
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.
> 
> 
> Source file was:
> #include <pthread.h>
> 
> void* test_func(void* data)
> {
>   return data;
> }
> 
> int main(void)
> {
>   pthread_t thread;
>   pthread_create(&thread, NULL, test_func, NULL);
>   pthread_detach(thread);
>   pthread_join(thread, NULL);
>   pthread_atfork(NULL, NULL, NULL);
>   pthread_exit(NULL);
> 
>   return 0;
> }
> 
> Determining if the function pthread_create exists in the pthreads failed with the following output:
> Change Dir: /home/VMuser/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeTmp
> 
> Run Build Command(s):/bin/ninja cmTC_21238 && [1/2] Building C object CMakeFiles/cmTC_21238.dir/CheckFunctionExists.c.o
> [2/2] Linking C executable cmTC_21238
> FAILED: cmTC_21238 
> : && /bin/cc -DCHECK_FUNCTION_EXISTS=pthread_create   CMakeFiles/cmTC_21238.dir/CheckFunctionExists.c.o  -o cmTC_21238  -lpthreads && :
> /bin/ld: cannot find -lpthreads
> collect2: error: ld returned 1 exit status
> ninja: build stopped: subcommand failed.

While the issue is similiar to #13759, re-downloading the gazebo plugins do not seem to make the difference.
Moreover,  running sudo apt-get install libprotobuf-dev libprotoc-dev protobuf-compiler libeigen3-dev libxml2-utils python-rospkg python-jinja2 throws the **error** (not sure if related to above error though) :

> Package python-rospkg is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> However the following packages replace it:
>   python3-rospkg
> 
> E: Package 'python-rospkg' has no installation candidate

Hence, I suspect the error could be Cmake files but I don't have the sufficient know-how to fix it.

Any advice is appreciated.

julianoes, 2022-01-12:
Try `sudo apt install python3-rospkg`.

Jeswin17, 2022-01-12:
> 

Tried it, it was already installed thus same error o/p :(

Jeswin17, 2022-01-12:
> Try `sudo apt install python3-rospkg`.

Tried it, it was already installed thus same error o/p :(

Jaeyoung-Lim, 2022-01-13:
@Jeswin17 I think your install script might have failed somewhere. Did the script output any errors?

Jeswin17, 2022-01-13:
@Jaeyoung-Lim  I do not think so.. 
[image]

Jeswin17, 2022-01-13:
Not sure if related but I am also not able to make jMAVSim, it gets stuck after the TCP port connection while throwing a ClassNotFoundExeption: org.apache.tools.ant.launch.Launcher.  I will open an issue for this also and link it here.

Jaeyoung-Lim, 2022-01-13:
@Jeswin17 It is related. You are missing some dependencies that should have been installed as part of the setup script.

Jeswin17, 2022-01-13:
How do I identify these missing dependencies since the setup?sh script does not throw any errors?

Jeswin17, 2022-01-13:
Looking at related issues with jAVSim, I tried using OpenJDK 11 & 8 and got the following errors.
**openjdk-11**
[image]

**openjdk-8**
[image]

julianoes, 2022-01-13:
Try a newer version.

Jeswin17, 2022-01-13:
@julianoes  no luck with it. Does openjdk need to bet set to JAVA_HOME in /etc/environmetnt file? Anyway I think I will try to downgrade to 18.4 and try it.

julianoes, 2022-01-13:
I suggest to use 20.04 and use what setup.sh suggests. If that does not work, we'll have to fix it.

julianoes, 2022-01-13:
On my Ubuntu 20.04 I have:
```
java --version
openjdk 13.0.7 2021-04-20
OpenJDK Runtime Environment (build 13.0.7+5-Ubuntu-0ubuntu120.04)
OpenJDK 64-Bit Server VM (build 13.0.7+5-Ubuntu-0ubuntu120.04, mixed mode, sharing)
```

And `make px4_sitl jmavsim` works.

Jeswin17, 2022-01-14:
@julianoes I was able to get jMAVSim to work! 
The issue was with environment variables JAVA_HOME & ANT_HOME which were not there in the environments. 

However the make gazebo still returns the original CMAKE/Booster related issue. Do you reckon it needs any environment variables?

julianoes, 2022-01-14:
@Jeswin17 please remove the build folder and try `make px4_sitl gazebo` again and paste the full output.

Jeswin17, 2022-01-14:
@julianoes  this is the output

[0/10] git submodule Tools/sitl_gazebo
[5/10] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /bin/cc
-- Check for working C compiler: /bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /bin/c++
-- Check for working CXX compiler: /bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.58") found components: system thread filesystem 
-- Found PkgConfig: /bin/pkg-config (found version "0.29.1") 
-- Checking for module 'bullet>=2.82'
--   Found bullet, version 2.88
-- Found ccd: /usr/include (found suitable version "2.0", minimum required is "2.0") 
-- Found fcl: /usr/include (found suitable version "0.5.0", minimum required is "0.3.2") 
-- Found assimp: /usr/include (found version "5.0.0") 
-- Found DART: /usr/include (Required is at least version "6.6") found components: dart 
-- Found Boost: /lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake (found suitable version "1.71.0", minimum required is "1.40.0") found components: thread system filesystem program_options regex iostreams date_time 
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found version "3.6.1") 
-- Looking for ignition-math6 -- found version 6.9.2
-- Searching for dependencies of ignition-math6
-- Looking for OGRE...
-- OGRE_PREFIX_WATCH changed.
-- Checking for module 'OGRE'
--   Found OGRE, version 1.9.0
-- Found Ogre Ghadamon (1.9.0)
-- Found OGRE: optimized;/usr/lib/x86_64-linux-gnu/libOgreMain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreMain.so
-- Looking for OGRE_Paging...
-- Found OGRE_Paging: optimized;/usr/lib/x86_64-linux-gnu/libOgrePaging.so;debug;/usr/lib/x86_64-linux-gnu/libOgrePaging.so
-- Looking for OGRE_Terrain...
-- Found OGRE_Terrain: optimized;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so;debug;/usr/lib/x86_64-linux-gnu/libOgreTerrain.so
-- Looking for OGRE_Property...
-- Found OGRE_Property: optimized;/usr/lib/x86_64-linux-gnu/libOgreProperty.so;debug;/usr/lib/x86_64-linux-gnu/libOgreProperty.so
-- Looking for OGRE_RTShaderSystem...
-- Found OGRE_RTShaderSystem: optimized;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so;debug;/usr/lib/x86_64-linux-gnu/libOgreRTShaderSystem.so
-- Looking for OGRE_Volume...
-- Found OGRE_Volume: optimized;/usr/lib/x86_64-linux-gnu/libOgreVolume.so;debug;/usr/lib/x86_64-linux-gnu/libOgreVolume.so
-- Looking for OGRE_Overlay...
-- Found OGRE_Overlay: optimized;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so;debug;/usr/lib/x86_64-linux-gnu/libOgreOverlay.so
-- Looking for ignition-math6 -- found version 6.9.2
-- Looking for ignition-transport8 -- found version 8.2.1
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/lib/x86_64-linux-gnu/libprotobuf.so;-lpthread (found suitable version "3.6.1", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.2
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Checking for module 'uuid'
--   Found uuid, version 2.34.0
[... 92 lines cut ...]
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
   used as include directory in directory /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo/unit_tests
[... cut ...]
See also "/home/VMadmin/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /home/VMadmin/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /home/VMadmin/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /home/VMadmin/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:235: px4_sitl] Error 1

julianoes, 2022-01-14:
I'm trying to reproduce this.

What's your output of `apt list --installed | grep boost`?

Jeswin17, 2022-01-14:
@julianoes 
WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libboost-all-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed]
libboost-atomic-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-atomic1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-atomic1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-chrono-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-chrono1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-chrono1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-container-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-container1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-container1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-context-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-context1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-context1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-coroutine-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-coroutine1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-coroutine1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-date-time-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-date-time1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-date-time1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-exception-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-exception1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-fiber-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-fiber1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-fiber1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-filesystem-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-filesystem1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-filesystem1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-graph-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-graph-parallel-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-graph-parallel1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-graph-parallel1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-graph1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-graph1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-iostreams-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-iostreams1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-iostreams1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-locale-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-locale1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-locale1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-log-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-log1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-log1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-math-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-math1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-math1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-mpi-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-mpi-python-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-mpi-python1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-mpi-python1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-mpi1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-mpi1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-numpy-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-numpy1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-numpy1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-program-options-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-program-options1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-program-options1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-python-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-python1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-python1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-random-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-random1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-random1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-regex-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-regex1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-regex1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
[... 2 lines cut ...]
libboost-serialization1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-stacktrace-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
[... cut ...]
libboost-type-erasure1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-type-erasure1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-wave-dev/focal,now 1.71.0.0ubuntu2 amd64 [installed,automatic]
libboost-wave1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost-wave1.71.0/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost1.71-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]
libboost1.71-tools-dev/focal,now 1.71.0-6ubuntu6 amd64 [installed,automatic]

Jaeyoung-Lim, 2022-01-14:
@Jeswin17 Are you running this inside a VM? I think this was never mentioned in the description

Jeswin17, 2022-01-14:
@Jaeyoung-Lim  VM!

Jeswin17, 2022-01-14:
@julianoes  So I freshly d/led the repo again and ran setup.sh 

Ubuntu 20.04

Installing PX4 general dependencies
Hit:1 http://sg.archive.ubuntu.com/ubuntu focal InRelease
Hit:2 http://sg.archive.ubuntu.com/ubuntu focal-updates InRelease
Hit:3 http://sg.archive.ubuntu.com/ubuntu focal-backports InRelease
Hit:4 http://packages.osrfoundation.org/gazebo/ubuntu-stable focal InRelease
Hit:5 http://security.ubuntu.com/ubuntu focal-security InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
ccache is already the newest version (3.7.7-1).
cmake is already the newest version (3.16.3-1ubuntu1).
file is already the newest version (1:5.38-4).
g++ is already the newest version (4:9.3.0-1ubuntu2).
gcc is already the newest version (4:9.3.0-1ubuntu2).
make is already the newest version (4.2.1-1.2).
python3 is already the newest version (3.8.2-0ubuntu2).
python3-dev is already the newest version (3.8.2-0ubuntu2).
python3-setuptools is already the newest version (45.2.0-1).
unzip is already the newest version (6.0-25ubuntu1).
zip is already the newest version (3.0-11build1).
astyle is already the newest version (3.1-2build1).
cppcheck is already the newest version (1.90-4build1).
lcov is already the newest version (1.14-2).
ninja-build is already the newest version (1.10.0-1build1).
python3-wheel is already the newest version (0.34.2-1).
shellcheck is already the newest version (0.7.0-2build2).
build-essential is already the newest version (12.8ubuntu1.1).
gdb is already the newest version (9.2-0ubuntu1~20.04).
git is already the newest version (1:2.25.1-1ubuntu3.2).
libxml2-dev is already the newest version (2.9.10+dfsg-5ubuntu0.20.04.1).
libxml2-utils is already the newest version (2.9.10+dfsg-5ubuntu0.20.04.1).
rsync is already the newest version (3.1.3-8ubuntu0.1).
python3-pip is already the newest version (20.0.2-5ubuntu1.6).
The following packages were automatically installed and are no longer required:
  libmbim-glib4 libmbim-proxy libqmi-glib5 libqmi-proxy usb-modeswitch
  usb-modeswitch-data
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.

Installing PX4 Python3 dependencies
Requirement already satisfied: argcomplete in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 1)) (2.0.0)
Requirement already satisfied: argparse>=1.2 in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 2)) (1.4.0)
Requirement already satisfied: cerberus in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 3)) (1.3.4)
Requirement already satisfied: coverage in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 4)) (6.2)
Requirement already satisfied: empy>=3.3 in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 5)) (3.3.4)
Requirement already satisfied: future in /usr/lib/python3/dist-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 6)) (0.18.2)
Requirement already satisfied: jinja2>=2.8 in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 7)) (3.0.3)
Requirement already satisfied: jsonschema in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 8)) (4.4.0)
Requirement already satisfied: kconfiglib in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 9)) (14.1.0)
Requirement already satisfied: lxml in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 10)) (4.7.1)
Requirement already satisfied: matplotlib>=3.0.* in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 11)) (3.5.1)
Requirement already satisfied: numpy>=1.13 in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 12)) (1.22.0)
Requirement already satisfied: nunavut>=1.1.0 in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 13)) (1.6.2)
Requirement already satisfied: packaging in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 14)) (21.3)
Requirement already satisfied: pandas>=0.21 in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 15)) (1.3.5)
Requirement already satisfied: pkgconfig in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 16)) (1.5.5)
Requirement already satisfied: psutil in ./.local/lib/python3.8/site-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 17)) (5.9.0)
Requirement already satisfied: pygments in /usr/lib/python3/dist-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 18)) (2.3.1)
Requirement already satisfied: wheel>=0.31.1 in /usr/lib/python3/dist-packages (from -r /home/VMadmin/PX4-Autopilot/Tools/setup/requirements.txt (line 19)) (0.34.2)
[... cut ...]
libimage-exiftool-perl is already the newest version (11.88-1ubuntu0.1).
gazebo11 is already the newest version (11.10.0-1~focal).
libgazebo11-dev is already the newest version (11.10.0-1~focal).
The following packages were automatically installed and are no longer required:
  libmbim-glib4 libmbim-proxy libqmi-glib5 libqmi-proxy usb-modeswitch
  usb-modeswitch-data
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 11 not upgraded.

Relogin or reboot computer before attempting to build NuttX targets

julianoes, 2022-01-14:
Looks all fine. And it still did not work afterwards? After a clean build?

CharlChc, 2022-02-15:
@julianoes when I run "make px4_sitl_default gazebo", I got the error report like this:
INFO  [dataman] Unknown restart, data manager file './dataman' size is 11798680 bytes
Gazebo multi-robot simulator, version 9.19.0
Copyright (C) 2012 Open Source Robotics Foundation.
Released under the Apache 2 License.
http://gazebosim.org

INFO  [simulator] Waiting for simulator to accept connection on TCP port 4560
[Msg] Waiting for master.
[Err] [Connection.cc:546] Connection[0] Closed during Read
[Err] [ConnectionManager.cc:135] Unable to read from master
[Err] [gazebo_shared.cc:77] Unable to initialize transport.
[Err] [gazebo_client.cc:56] Unable to setup Gazebo
^CERROR [px4] Startup script returned with return value: 2

Seems like gazebo cannot be launched. Could you please let me know how to solve it. thanks in advance.

julianoes, 2022-02-19:
@CharlChc your issue looks different from this thread. I suggest you open a new issue and add some more information, e.g. which OS and which OS version, etc.

channing-prince, 2022-05-14:
Hi, I have the same problem, but now i solved it !!!
Well, not exactly solved, i just used a new Ubuntu 20.04 (ROS noetic installed) ， and it passed the compile

I have reason to believe that the cartographer or the dependencies caused this problems , I tried to 'make px4_sitl gazebo'  on two Ubuntu system and both failed , both of them are configured with cartographer, and cartographer all works fine (used with rplidar, haven‘t try simulation before) 
so  today , I  try the ubuntu only installed ros and it worked !!!  I don't know where is the problem, I would be glad if this can help you to  reproduce  or solve it

Tips：
all above are on the vm
Ubuntu 20.04
ROS Noetic
Cartographer ROS follow the [instruction ](https://google-cartographer-ros.readthedocs.io/en/latest/compilation.html)

julianoes, 2022-05-15:
Looks like the apt sources and dependencies were in a bad state. I'm glad a re-install fixed it.

closed 2022-05-15 by julianoes

eastman-bechamo, 2022-06-16:
@CharlChc Did you ever open a new issue?

fbenti, 2022-12-19:
> 

I had the same problem of @Jeswin17 in https://github.com/PX4/PX4-Autopilot/issues/18999#issuecomment-1012835041, solved with https://github.com/PX4/PX4-Autopilot/issues/13759#issuecomment-623235991 changing python to python3