PX4/PX4-Autopilot #19005: PX4 Subscribes to Trajectory Setpoint from Offboard Mode During RTL

adk4davsys, 2022-01-13:
**Describe the bug**
Triggering RTL when publishing to `trajectory_setpoint` topic in `Offboard Mode` vehicle continues to subscribe to the topic and makes vehicle unstable.

**To Reproduce**
Steps to reproduce the behavior:
1. Publish to `/fmu/trajectory_setpoint/in` within `Offboard Mode`
2. Trigger RTL
3. Vehicle attempts to RTL while still setting position setpoint using the topic `/fmu/trajectory_setpoint/in`
4. Eventually vehicle returns to home but does not disarm and continues to point toward `/fmu/trajectory_setpoint/in`

**Expected behavior**
My understanding is that triggering RTL should ignore `/fmu/trajectory_setpoint/in` from `Offboard Mode`. Shown below are the comparison between publishing to `/fmu/trajectory_setpoint/in` and not publishing to the topic during RTL.

**Log Files and Screenshots**
[image]
[image]

**Drone (please complete the following information):**
- Based on V1.12.3
- Pixhawk 4

**Additional Context
This behavior was identified from issue #18983

benjinne, 2022-01-15:
Currently there's a bug with `trajectory_setpoint` being published by both offboard and `mc_pos_control` I fixed it in this PR https://github.com/PX4/PX4-Autopilot/pull/18318 but it's still waiting review

adk4davsys, 2022-01-17:
@benjinne Thank you for following up on this issue. For the time being, I only publish to the topic while the copter is in `offboard mode`.