PX4/PX4-Autopilot #19009: Bug of "make px4_sitl gazebo"

Marwa-ABDELOUINISSE, 2022-01-14:
(base)user:~/PX4/PX4-Autopilot$ sudo make px4_sitl gazebo
[0/5] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
CMake Error at /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:117 (find_package):
  Could not find a package configuration file provided by "boost_filesystem"
  (requested version 1.71.0) with any of the following names:

    boost_filesystemConfig.cmake
    boost_filesystem-config.cmake

  Add the installation prefix of "boost_filesystem" to CMAKE_PREFIX_PATH or
  set "boost_filesystem_DIR" to a directory containing one of the above
  files.  If "boost_filesystem" provides a separate development package or
  SDK, be sure it has been installed.
Call Stack (most recent call first):
  /usr/lib/x86_64-linux-gnu/cmake/Boost-1.71.0/BoostConfig.cmake:182 (boost_find_component)
  /usr/share/cmake-3.16/Modules/FindBoost.cmake:443 (find_package)
  CMakeLists.txt:47 (find_package)

-- Configuring incomplete, errors occurred!
See also "/volatile/home/user/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeOutput.log".
See also "/volatile/home/user/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo/CMakeFiles/CMakeError.log".
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-configure 
cd /volatile/home/user/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/bin/cmake -DCMAKE_INSTALL_PREFIX=/usr/local -DSEND_ODOMETRY_DATA=ON -DGENERATE_ROS_MODELS=ON -GNinja /volatile/home/user/PX4/PX4-Autopilot/Tools/sitl_gazebo && /usr/bin/cmake -E touch /volatile/home/user/PX4/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-configure
ninja: build stopped: subcommand failed.
make: *** [Makefile:235 : px4_sitl] Erreur 1

julianoes, 2022-01-14:
Looks somewhat related to https://github.com/PX4/PX4-Autopilot/issues/18999.

khanasif786, 2023-03-23:
Use:
```sudo apt-get install libgstreamer-plugins-base1.0-dev gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly -y```

julianoes, 2023-03-23:
Closing, this should work fine when the ubuntu setup script is used: Tools/setup/ubuntu.sh

abhinav-mzn, 2023-05-17:
@khanasif786 Thanks, It worked