PX4/PX4-Autopilot #19014: Ainstein US-D1 not working

msnesbit, 2022-01-14:
Current build 1.12.3 on a Pixhawk 4
The sens_uland_cfg exists in the parameter list which, as I understand, means that the driver exists on this firmware build.
I have tried every port option I can think of, as well as supplying power to the radar via a BEC and not from the Pixhawk power rail,  but I have yet to read any data from it. The radar is getting power and pulling a very small amount of current which indicates to me that the unit is likely functioning. Any help would be much appreciated!

dagar, 2022-01-14:
Unfortunately I don't have one of these units, but we can try to debug remotely.

Through a console (mavlink or serial console) can you run `dmesg`, `perf`, and `work_queue status`? I'm trying to see if it's failing with any useful output.

msnesbit, 2022-01-14:
Here you go. 

[image]
[image]
[image]
[image]

FZenjoys, 2022-11-15:
@msnesbit Hello, did you solve that problem? I guess I face the same one.
The data output of US D1 is working through its tool test software (see the figure below). Unfortunately, there is no DISTANCE_SENSOR in the MAVLink Inspector after the guide configuration (SENS_ULAND_CFG --> TELEM 2). (power 5.06V, correct RX & TX )
By the way, the TFmini plus is working well via a similar configuration.
 
[image]

martacportugal, 2023-04-30:
> @msnesbit Hello, did you solve that problem? I guess I face the same one. The data output of US D1 is working through its tool test software (see the figure below). Unfortunately, there is no [...]

Hello! :) Did you manage to solve this? I have the same problem.

msnesbit, 2023-05-03:
Yeah I did end up solving this. The solution is sitting somewhere on an old
email account. Let me dig it up. If I recall, you need to edit one line in
the firmware... easy fix.

martacportugal, 2023-05-09:
Let me know if you found it! 
I tried adding  'CONFIG_DRIVERS_DISTANCE_SENSOR_ULANDING_RADAR=y' to the default.px4board but it doesn't seem to be working.