PX4/PX4-Autopilot #19016: Full Implementation of Newton-Euler Dynamics

ErcBunny, 2022-01-15:
Hello!

I have been trying to modify PX4 code for my own over-actuated hexacopter. To be more specific, I want to 
implement the algorithm described in this [paper](https://arxiv.org/abs/1810.06258?utm_source=feedburner&utm_medium=feed&utm_campaign=Feed%253A+arxiv%252FQSXk+%2528ExcitingAds%2521+cs+updates+on+arXiv.org%2529).

[image]

 It seems that the second row is already implemented largely in `AngularVelocityControl.cpp`. I added the center of mass offset term without difficulty. But I ran into problems when adding the `body angular velocity` cross `body linear velocity` term to the first row of equation.

I tried to add that term in `AngularVelocityController.cpp` where variable `_thrust_sp` is updated and scaled in Newton, by adding two subscriptions to `vehicle_local_position` (so that I can get velocity in NED) and `vehicle_attitude` (so that I can rotate velocity into body frame) topics. Body angular velocity is inherently updated in this file.

I tested this implementation in SITL with a custom tilt-rotor hex (just like the one described in the aforementioned paper), vehicle performance got worse when the cross term was added. I have no idea why this is happening. So, I am writing this issue to seek help.

bresch, 2022-01-17:
Hi @ErcBunny , can you share the modifications you've made? Maybe we could spot a mistake. Without that, it's quite difficult to guess. 

- Maybe you forgot to multiply it by `m`? (just saying that because I thought the `m` was missing at first)
- NED velocity, attitude and angular rates have different update frequencies and won't be available at each update cycle together, did you consider that in you implementation?

ErcBunny, 2022-01-17:
> Hi @ErcBunny , can you share the modifications you've made? Maybe we could spot a mistake. Without that, it's quite difficult to guess.
>
> * Maybe you forgot to multiply it by `m`? (just saying that [...]

Hello @bresch , thanks for your attention.

1. Please kindly take a look at my forked repo [here](https://github.com/ErcBunny/PX4-Autopilot/tree/omniHex). To be more specific, please go to [line 181](https://github.com/ErcBunny/PX4-Autopilot/commit/28ef321c6889b3ec20476e34d3ac05a5a8bd3ae2#diff-57b8248090f20f5df0e3c5fa29d9fa615509a0471d94b9da27400d3aa27eb4e5) in `AngularVelocityController.cpp`. 
2. I did think about the update frequency problem. But I have no idea why that is going to be causing instability... If this is indeed the problem, I don't know how to further modify the code. So in my implementation, it is not yet considered.
3. Since there are additional ros2 nodes running together with sitl, if you decide that you need to run the full SITL simulation to better identify the problem, please let me know.

Regards~

bresch, 2022-01-19:
@ErcBunny I think the problem is that this term (coriolis compensation, right?) is doing too much. In fact, if you look [here](https://github.com/ErcBunny/PX4-Autopilot/commit/28ef321c6889b3ec20476e34d3ac05a5a8bd3ae2#diff-57b8248090f20f5df0e3c5fa29d9fa615509a0471d94b9da27400d3aa27eb4e5R195), the normalized thrust setpoint is converted to a -non-normalized- one using the mass of the vehicle and the normalized hover thrust. However, the compensation you're adding is already a non-normalized force, so it should be added after this line.

Hope that helped

ErcBunny, 2022-01-22:
@bresch I tried to add this term on the scaled `_thrust_sp` and it worked! Thank you for the help.

closed 2022-01-22 by ErcBunny

ErcBunny, 2024-02-11:
The cross product term should only be added if the acceleration setpoint is expressed in the body frame, which is not the case in the paper, but not the case in PX4. So this term should not be added.