PX4/PX4-Autopilot #19017: ros2 how to publish vehicle_trajectory_waypoint topic?

xiaobai-cai-hold, 2022-01-16:
I want to publish vehicle_trajectory_waypoint topic in px4_ros_com/offboardcontrol.cpp to contrlo vehicle to fly several points or make the vehicle to work on mission mode. I look up px4_msgs/msg/VehicleTrajectoryWaypoint.msg, it includes the following content

uint64 timestamp                # time since system start (microseconds)

uint8 MAV_TRAJECTORY_REPRESENTATION_WAYPOINTS = 0

uint8 type # Type from MAV_TRAJECTORY_REPRESENTATION enum.

uint8 POINT_0 = 0
uint8 POINT_1 = 1
uint8 POINT_2 = 2
uint8 POINT_3 = 3
uint8 POINT_4 = 4

uint8 NUMBER_POINTS = 5

TrajectoryWaypoint[5] waypoints

I dont't know how to set waypoints' value and type values. I find some instruction that can study
https://mavlink.io/en/messages/common.html#TRAJECTORY_REPRESENTATION_WAYPOINTS
https://mavlink.io/en/messages/common.html#MAV_CMD
https://mavlink.io/en/services/trajectory.html

I don't know how to set waypoints values , can someone help me solve this problem? thanks

tank104, 2022-08-12:
Did you manage to sort this? I too would like to upload missions via ros2

xiaobai-cai-hold, 2022-08-20:
>sorry,I cant solve it.

closed 2025-11-16 by github-actions[bot]