PX4/PX4-Autopilot #19031: EKF2 doesn't initialize without magnetometer

okalachev, 2022-01-18:
If the magnetometer is disabled with `SYS_HAS_MAG` = 0 and `EKF2_MAG_TYPE` = None, then EKF2 doesn't initialize the local position when gets the data from (external) optical flow.

```
nsh> ekf2 status
ekf2:0 attitude: 1, local position: 0, global position: 0
```

This makes autonomous flight impossible.

In certain cases we should disable the magnetometer (or a flight controller may not have the magnetometer at all, which is legit, I think). Performing position and autonomous flight using gyroscope + flow is totally OK, but how to do that?

dagar, 2022-01-18:
What do you think @bresch?

lukegluke, 2022-01-19:
Wasn't it fixed by https://github.com/PX4/PX4-ECL/pull/931 already? Or something broke that?

okalachev, 2022-01-19:
I check on the latest version v1.12.3.

bresch, 2022-01-19:
> Wasn't it fixed by PX4/PX4-ECL#931 already? Or something broke that?

That PR was to let the EKF initialize without magnetometer, which it does; I just checked with current master. The title of that issue is a bit misleading, it's not about initializing (the EKF runs and provides an attitude estimate), but using the optical flow.

@okalachev Yes, we should be able to fly in position controlled mode using an optical flow sensor without mag. The starting condition checks for the yaw alignment here: https://github.com/PX4/PX4-Autopilot/blob/632dfa55e6f231f1b62fb2e9bfb048d391ebafc9/src/modules/ekf2/EKF/control.cpp#L453-L454
Since there are no sources of heading, the yaw is never aligned. However, we might just want to add an alternate condition to the yaw alignment using the mag type parameter. I didn't check if some other part of the optical flow fusion would fail without the yaw align flag being true, but it could be just enough.

So simply try adding `_params.mag_fusion_type == MAG_FUSE_TYPE_NONE` to that condition. If it doesn't work, the quickest way to debug it would be to create a unit test with that configuration or by using SITL (iris opt flow model).

okalachev, 2022-01-19:
@bresch, hi, seems as your change has fixed the issue!

bresch, 2022-01-19:
@okalachev Cool, could you make a PR please?

okalachev, 2022-01-19:
@bresch, OK.

okalachev, 2022-01-20:
@bendyer, #19052.

closed 2022-01-27 by bresch