PX4/PX4-Autopilot #19044: Multi-Vehicle Simulation with Gazebo and ROS

pitergit95, 2022-01-19:
Hi,

I'm trying to add new iris drone to my multi-vehicle simulation in Gazebo with ROS but i cannot find in latest Firmware the file which is pointed in documentation. It should be here(path): PX4-Autopilot/posix-configs/SITL/init/test/. I can find this file in oldest branch from 2018 but in the latest it is missing. There could be a mismatch between documentation and latest branch. Could you please help me and point where i can add new drone to simulation?

Thank you in advance

Jaeyoung-Lim, 2022-02-08:
@pitergit95 Which documentation are you referring to?

pitergit95, 2022-02-17:
Yes, that was the path which i was searching for file to add new drone. Still i cannot find file mentioned here in newest firmware release: 
_create a startup file, and change the file as follows:
make a copy of an existing iris rcS startup file (iris_1 or iris_2) and rename it iris_3_
Where i can find this rcS file?