PX4/PX4-Autopilot #19046: X500V2 kit Jumps when switching to AltHold

farhangnaderi, 2022-01-19:
After trying to fly with the Holybro X500 V2 kit which flies well with Stabilized mode, I noticed that the drone jumps up and down when switching to Altitude mode. The kit is equipped with Pixhawk 5X. A log is also as below:

https://review.px4.io/plot_app?log=21beb351-f2bb-4767-b71f-725946098ec0

I found something which might be relevant [here](https://github.com/PX4/PX4-Autopilot/issues/11717) but not sure if it is the case!!!

dagar, 2022-01-19:
Can you share a picture of this vehicle?

farhangnaderi, 2022-01-19:
Sure, here you go: 

[image]

If there is a possible question of how I have mounted the Pixhawk: 

[image]

dagar, 2022-01-19:
Can you try reducing the hover thrust quite a bit? `MPC_THR_HOVER 0.23`

Did you calibrate the ESCs?

bresch, 2022-01-20:
Not sure if it's related, but both times before the spike in alt mode, the `local_position_setpoint.vz` is missing a couple of samples:
[image]

dagar, 2022-01-20:
It's the land detected ground contact.
[image]

dagar, 2022-01-20:
The misconfigured `MPC_THR_HOVER` could probably also be blamed for the land detector issues (leading to throttle spikes).

@MaEtUgR could we ramp up to this more gradually? https://github.com/PX4/PX4-Autopilot/blob/a2260e53da48495644e128562ad80c1c276e7bef/src/modules/mc_pos_control/MulticopterPositionControl.cpp#L429

MaEtUgR, 2022-02-09:
> The misconfigured MPC_THR_HOVER could probably also be blamed for the land detector issues (leading to throttle spikes).

Right, you saw that from the has_low_thrust flag being constantly true and hence it getting into ground contact all the time:
[image]

MaEtUgR, 2022-02-09:
From the data it looks like either the drone is completely overpowered/"overgeared" but since the motors, ESCs and props reportedly stayed the same that's unlikely. What seems more likely is a problem with the output scaling. So either ESC calibration gone wrong. The ESC calibration was reportedly done so that might be the issue or the PWM range configuration. If the ESCs support DSHOT and that gets enabled it would avoid such issues. Otherwise a fresh 1000 - 2000us PWM calibration on the ESCs if necessary with the motor test would solve that part and PWM parameter defaults the range configuration.

farhangnaderi, 2022-02-15:
@MaEtUgR @bresch Thanks for the review, it might be my radio since I have a three-select for flight modes and Altitude is before Position mode exactly. 
@dagar Hover thrust less than 40% was really lazy to takeoff as I understood from today's flight test. 

I have calibrated ESCs again and flew today, these are the logs: 
https://review.px4.io/plot_app?log=50fac13b-cf22-4259-9dca-ceb6f916e17f
https://review.px4.io/plot_app?log=f2b50c49-a5e5-4aea-ae2d-17c74aa0a03a
https://logs.px4.io/plot_app?log=4bd7236b-c02f-4756-8dab-fa1acd4455f0
https://logs.px4.io/plot_app?log=f7b08d90-f681-49e5-a52f-feaecaf0ef27
https://review.px4.io/plot_app?log=6827a6d8-43cb-42b1-8f2a-107d5065d7a5