PX4/PX4-Autopilot #19049: I keep having this issue. please help me on this.

curius132, 2022-01-20:
d@b:~/px4/Firmware$make px4_sitl gazebo_standard_vtol
[1/849] git submodule src/drivers/gps/devices
[5/849] git submodule src/modules/mavlink/mavlink
[10/849] git submodule Tools/sitl_gazebo
[14/849] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Traceback (most recent call last):
  File "/home/d/px4/Firmware/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py", line 16, in <module>
    from pymavlink.generator import mavgen
  File "/home/d/px4/Firmware/src/modules/mavlink/mavlink/pymavlink/generator/mavgen.py", line 26, in <module>
    from future import standard_library
ModuleNotFoundError: No module named 'future'
[16/849] No download step for 'sitl_gazebo'
FAILED: mavlink/development/development.h 
cd /home/d/px4/Firmware/build/px4_sitl_default/src/modules/mavlink && /usr/bin/python3 /home/d/px4/Firmware/src/modules/mavlink/mavlink/pymavlink/tools/mavgen.py --lang C --wire-protocol 2.0 --output /home/d/px4/Firmware/build/px4_sitl_default/mavlink /home/d/px4/Firmware/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
[21/849] Generating uORB topic headers
ninja: build stopped: subcommand failed.
Makefile:235: recipe for target 'px4_sitl' failed
make: *** [px4_sitl] Error 1

Fer-Duarte, 2022-08-05:
I have the same error. Hope someone can help us.

thatgirlAm, 2023-02-04:
lol same...I was hoping to find responses...