PX4/PX4-Autopilot #19050: Modifying (or not) trim airspeed on climbout?

tstastny, 2022-01-20:
In #18810 I made this change:
https://github.com/PX4/PX4-Autopilot/commit/082e32019183d6fe1c57c75194816d2053fc1899#diff-fb4799f616fdcae17e69051258ca53340f3e41b706805d5f86d17a61731e4feaL1414-L1434
^FYI you'll need to click "load diff" on the FixedwingPositionControl.cpp to see the lines that are linked.. it will auto scroll you to that point once the diff is shown.

My thought was, why would we want only trim airspeed passed directly here during climbout when we otherwise use get_auto_airspeed_setpoint() which can increase the airspeed setpoint for safety considering e.g. wind or high roll?

After discussion with @sfuhrer, thought it may be nice to pose this question to anyone else who may have a specific reason this was set that way (and change it back if necessary).

@dagar @RomanBapst (pls feel free to tag anyone else who may have an opinion on this)

sfuhrer, 2022-01-20:
@Antiheavy @ryanjAA

ryanjAA, 2022-01-30:
What’s this do? Try and achieve trim speed on climbout or adjust trim airspeed on climbout to take wind and roll (so increased stall speed) into account?

tstastny, 2022-01-30:
> What’s this do?

@ryanjAA, the latter. Previously, wind and roll (stall speed) were not considered on climbout. As considering these effects will only cause increased airspeed setpoints.. things should only get safer.

However, I wanted to put it out to the community to see if there was any specific reason why this was not set this way before (other than maybe just an oversight). Instead of silently changing behavior without notice to any fixed-wing users.

ryanjAA, 2022-01-30:
Thanks @tstastny for the explanation. 

It was probably just an oversight or simplification. I say do it and make the change. There really shouldn’t be an issue with increasing demanded speed other than more battery usage but it would be good in tail wind situations and does seem safer.  It does work though as it currently is so there is a KISS approach that should be considered but I can’t see why making it safer could hurt. 

One caveat though, this should be one way, I.e. there should be something where it can’t lower speed (maybe that’s already in there, not min airspeed, just in general on climbout) so it can increase climbout airspeed but not lower it. I’m thinking of scenarios where you’re climbing and there’s something wrong with airspeed and wind reading are wrong, etc. 

I wonder how this is handled in full-size aircraft, procedurally for comparison but I like the added safety aspect here.

tstastny, 2022-02-01:
thanks @ryanjAA for your feedback.
> there should be something where it can’t lower speed 

this is how it is. airspeed cannot be reduced below trim in this state. only increased if safety demands

since it's been two weeks - i'm closing this. if anyone else has different opinions, please feel free to re-open!

closed 2022-02-01 by tstastny