PX4/PX4-Autopilot #19053: checkVerticalAccelerationHealth() failing while on the ground

ashwinvk94, 2022-01-20:
**Describe the bug**
During some flights, I noticed that [checkVerticalAccelerationHealth()](https://github.com/PX4/PX4-Autopilot/blob/a2260e53da48495644e128562ad80c1c276e7bef/src/modules/ekf2/EKF/control.cpp#L776) is failing continuously when the drone is on the ground. I also figured out that it is because is[_clipping_frequently and is_inertial_nav_falling](https://github.com/PX4/PX4-Autopilot/blob/a2260e53da48495644e128562ad80c1c276e7bef/src/modules/ekf2/EKF/control.cpp#L819) is true. This causes the the [height to get reset](https://github.com/PX4/PX4-Autopilot/blob/a2260e53da48495644e128562ad80c1c276e7bef/src/modules/ekf2/EKF/control.cpp#L646) every time when its on the ground.

**Is checkVerticalAccelerationHealth() required when only using the rangefinder for height estimates?**

In the below logs files, if you check vehicle_local_position.z_reset_counter, you'll see that it getting reset continously prior to takeoff.

**Log Files**
https://review.px4.io/plot_app?log=787bc24c-9295-406f-a8ff-1c6125496e70
https://review.px4.io/plot_app?log=5f290c66-70ef-4d19-9a19-450d5933fa6a

dagar, 2022-01-20:
Starting to look at your logs. 

I see you have EKF2_EV_DELAY set to 800 milliseconds which seems enormous. Is that correct?

dagar, 2022-01-20:
Why are your external vision vertical velocity innovation variance and test ratio 0 the entire time?

[image]

ashwinvk94, 2022-01-20:
We are **NOT fusing the vertical velocity from the vehicle_visual_odometry msg.** The data on this topic is coming from velocity estimates from a 2D lidar. And hence we don't have vertical velocity estimates from the sensor.

**I have [commented out](https://github.com/PX4/PX4-ECL/blob/master/EKF/control.cpp#L330) the vertical velocity fusion in the firmware to NOT fuse the vertical velocity estimates.**

Yeah. EKF2_EV_DELAY is 800ms is correct. We are working on reducing that. But the estimator still fuses the horizontal velocity data well.

ashwinvk94, 2022-01-20:
Could you point me to converstations when checkVerticalAccelerationHealth() was implemented??

closed 2025-11-15 by github-actions[bot]