PX4/PX4-Autopilot #19074: Multi-UAV launch using single xacro file

NishanthARao, 2022-01-26:
Greetings all.

I have a complex system that consists of 4 UAVs that are attached to a single payload via rigid links. Thus, I have written a SINGLE .xacro file that contains the description of the 4 UAVs with payload and link connections. For your reference, it looks similar to this [example](https://github.com/intelligent-control-lab/Collaborative_Aerial_Transportation#launch-an-example). I need to use mavros and QGroundControl to talk to these 4 drones via mavlink. 

I saw the multi_uav launch file provided by default, but that launches single UAV files in seperate namespace. This doesn't help in my case, as I have a single .xacro file for all the 4 UAVs. How can I connect the mavros nodes and get mavros topics (in seperate namespaces) for all the 4 drones?

Thanks in advance. I would be grateful for your help and suggestions.

Jaeyoung-Lim, 2022-01-26:
@NishanthARao You can directly connect to the udp ports defined in the mavlink interface: https://github.com/PX4/PX4-SITL_gazebo/blob/2750fe233c6a38dbe60e545ac0ec878d6ac20b1e/models/iris/iris.sdf.jinja#L454

Since urdf does not support the latest SDF syntaxes, we have deprecated xacro from the PX4 tool chain and am now using jinja templates to template the sdf files. You can generate sdf files from urdf files that are generated from xacro files.

NishanthARao, 2022-01-27:
I see. So I need to specify 4 different mavros interfaces in the same sdf file? But how do I attach the interfaces to 4 seperate UAVs? Like there is no field for the link name to which the mavros interface is supposed to be connected. Say, I have 4 links that are named as UAV1, UAV2, UAV3 and UAV4. How can I specify this in the mavros interface, as I need to give seperate UDP port numbers to these links.

Jaeyoung-Lim, 2022-01-27:
> 1) But how do I attach the interfaces to 4 seperate UAVs? Like there is no field for the link name to which the mavros interface is supposed to be connected. Say, I have 4 links that are named as UAV1, UAV2, UAV3 and UAV4. How can I specify this in the mavros interface, as I need to give seperate UDP port numbers to these links.

mavros does not attach to a link, but a communication interface e.g. a UDP port. You can pass the address of the UDP port through fcu_url : https://github.com/mavlink/mavros/blob/b0297e39947a4dd6f60213370c864b9139254b37/mavros/launch/px4.launch#L5

> 2) Moreover, the example file that you have provided has 8 rotors. But I am interested in only Quad rotor system. Should I define it only for 4 rotor links?

It has four rotors

NishanthARao, 2022-01-27:
@Jaeyoung-Lim I'm sorry if the earlier post isn't clearly described. My thoughts are as follows.

1) When you say that you need to specify a communication interface, like a UDP port, It must distinguish between the four UAVs right? Like for eg., UDP1 - UAV1 and so on to UDP4 - UAV4. Now, my xacro file (or sdf whichever) has 4 links that describe the UAV. In addition to this, it also has links that correspond to a common payload that the UAVs lift, and the 4 rigid links that connect each UAV to the payload. If I were to simply provide for different communication interface macros (like the one shown in the example you have provided), how will the mavros topics eventually find the UAV links? Like /UAV0/mavros/global_position/global topic gives GPS values. How will this topic find the GPS position of the UAV if there is no link parameter in the mavros interface? Please remember that in my case, I have the entire system (all the 4 UAVs and the payload) described in a SINGLE xacro (or sdf) file. 

2) In [https://github.com/PX4/PX4-SITL_gazebo/blob/2750fe233c6a38dbe60e545ac0ec878d6ac20b1e/models/iris/iris.sdf.jinja#L454](this) link that you provided earlier, there are eight rotor channels inside the macro ```<control_channels>```. I think I have to provide only 4 for a quadrotor.

Jaeyoung-Lim, 2022-01-27:
>  How will this topic find the GPS position of the UAV if there is no link parameter in the mavros interface?

Mavros does not have any knowledge whether your vehicle is a simulated vehicle or a real vehicle and is just communicating through the udp port. The information of the sensors are passed through a gazebo transport, using the mavlink interface. For your case, you need to write the sdf file so that each sensors can keep track of each of the submodels, and then your rigid link between them keep attach all the submodels in some form.

If you are not sure how to attach gazebo plugins on sdf/urdf files, you can follow tutorials such as https://gazebosim.org/tutorials?tut=ros_gzplugins

> In https://github.com/PX4/PX4-SITL_gazebo/blob/2750fe233c6a38dbe60e545ac0ec878d6ac20b1e/models/iris/iris.sdf.jinja#L454 link that you provided earlier, there are eight rotor channels inside the macro <control_channels>. I think I have to provide only 4 for a quadrotor.

The iris model is a quadrotor. Please try running the model.

NishanthARao, 2022-01-27:
Thank you for your guidance on this and your quick response. Further, it is surprising to see that there are 8 rotor channels for a quadrotor which doesn't make any sense. 

I will get back to this issue once I figure out the UDP port connections. Once again, @Jaeyoung-Lim Thanks a lot for your time and info!

closed 2022-01-31 by Jaeyoung-Lim

NishanthARao, 2022-05-29:
@Jaeyoung-Lim I have written a single sdf file that contains two iris (in the same .sdf file). How can I proceed to launch this in Gazebo with mavros and PX4 SITL?