PX4/PX4-Autopilot #19077: offboard mode for fixedwing

airsimdevdd, 2022-01-27:
why  offboard mode for fixedwing is not supported in v1.12.3 and  but supported in v1.11, Iam using mavsdk offboard control

Jaeyoung-Lim, 2022-01-27:
They are supported in both versions

airsimdevdd, 2022-01-27:
but i can't fly in offboard using latest 1.12.3 same code works for 1.11 ,i can't understand why ? any solution

Jaeyoung-Lim, 2022-01-27:
I am currently using offboard on a daily basis, therefore the interface is functional.

I think you are clearly missing something but with the information you have given it is hard to tell what. - That is why we have issue templates

Please provide more specifics on what you are doing it how you are doing it and how it is failing. A  step by step instruction on how to repdroduce your problem would be helpful on spotting what the problem is

airsimdevdd, 2022-01-27:
Im using fixed wing sitl in gazebo using mavsdk- python previously i used the offboard_attitude.py with px4 v1.11 ,my fw mode changes to offboard and takeoff and complete waypoints but when i upgraded to latest version px4 1.12.3 my stays down like armed and does nothing doesnot change the 
mode

Jaeyoung-Lim, 2022-01-27:
> Im using fixed wing sitl in gazebo using mavsdk- python, previously when i used the offboard_attitude.py with px4 v1.11 ,my fw mode changes to offboard and takeoff and complete waypoints but when i upgraded to latest version px4 1.12.3 my fw stays down like armed and does nothing doesnot change the mode

@airsimdevdd I am a bit confused from what you have described. If you are using attitude setpoints you are controlling the attitude of the vehicle and therefore you won't be able to do waypoint following. What are you control?

airsimdevdd, 2022-01-28:
yes sorry it's setpoints

Jaeyoung-Lim, 2022-01-28:
@airsimdevdd Which setpoint are you sending?

airsimdevdd, 2022-01-28:
https://review.px4.io/plot_app?log=dbbbb1e8-dd46-4071-b5e2-80a1e425ebd9 log of 1.11

airsimdevdd, 2022-01-28:
https://review.px4.io/plot_app?log=ada2da4a-7890-4030-9c61-afc9e68a5d7c same code vehicle not responding in 1.12.3

Jaeyoung-Lim, 2022-01-28:
@airsimdevdd Please answer the question: "Which setpoint are you sending"

airsimdevdd, 2022-01-28:
im sending the attitude setpoints of pitch ,thrust and roll

airsimdevdd, 2022-01-28:
enabling virtual joystick solved the problem thanks alot

closed 2022-01-28 by airsimdevdd