PX4/PX4-Autopilot #19078: velocity setpoints for fixedwing

airsimdevdd, 2022-01-27:
Is there any way  to set velocity setpoints to fixedwing ,mavsdk velocity setpoints are not working i dont know why ?

Jaeyoung-Lim, 2022-01-27:
@airsimdevdd Support for velocity setpoints for fixedwing was only added in https://github.com/PX4/PX4-Autopilot/pull/18495

This is why it is not written in the documentation (yet): https://docs.px4.io/master/en/flight_modes/offboard.html#fixed-wing

airsimdevdd, 2022-01-27:
can use the now anyway to use it

Jaeyoung-Lim, 2022-01-27:
@airsimdevdd yes it is merged in latest master

airsimdevdd, 2022-01-27:
But , Iam  unable  to set velocity setpoints using the latest master

airsimdevdd, 2022-01-28:
[](url)https://review.px4.io/plot_app?log=96f7c7b5-6856-4c84-9434-f8368fdf5773

Jaeyoung-Lim, 2022-01-28:
@airsimdevdd The vehicle never went into offboard mode. I think the log is irrelevant to verify the functionality of offboard mode

airsimdevdd, 2022-01-28:
why it is not going to offboard

Jaeyoung-Lim, 2022-01-28:
@airsimdevdd It says in the log, that since you don't have a manual stick input, a failsafe is triggered. Try enabling virtual joystick in QGC and retry

airsimdevdd, 2022-01-28:
thanks vehicle went to offboard but the velocity doesnot  chnage according my setpoints

Jaeyoung-Lim, 2022-01-28:
@airsimdevdd Only the bearing vector and climbrates are set with velocity setpoints since you don't want to change your air relative velocity for fixedwing vehicles.

airsimdevdd, 2022-01-28:
we want to change the airspeed can't we do that

Jaeyoung-Lim, 2022-01-28:
@airsimdevdd Yes, you cannot change the airspeed

airsimdevdd, 2022-01-28:
By anyway  can i control the airspeed using atittude setpoints or any 
other way to control airspeed in offboardmode

Jaeyoung-Lim, 2022-01-28:
@airsimdevdd Currently this is not supported. If you are interested in extending the offboard interface to control airspeed, this would be very welcome

closed 2022-01-29 by Jaeyoung-Lim