PX4/PX4-Autopilot #19081: MicroRTPS bridget send rate

dsix-ls2n, 2022-01-27:
**Describe problem solved by the proposed feature**
With the current micro RTPS bridge, it is not possible to set the rate at which the topics are broadcast on the used port. With serial bandwidth, it can be an issue if too many topics are broadcast. However, some topics cast to ROS2 might not be necessary at the rate they are available on autopilot.

**Describe your preferred solution**
Include a rate option in the yaml configuration file to choose the ratio between the uorb topic publication and the broadcast through the RTPS bridge.

If this feature is not in progress and has some interest for the community. I might provide a PR for it.

benjinne, 2022-02-08:
This might be what you're looking for https://github.com/PX4/PX4-Autopilot/pull/18446

dsix-ls2n, 2022-02-08:
Thank you for this information. This is a good start, but I would like to be able to specify each topic rate specifically. Because depending on the application, some topics might be needed at full rate, but some others might not.

TSC21, 2022-02-08:
> Thank you for this information. This is a good start, but I would like to be able to specify each topic rate specifically. Because depending on the application, some topics might be needed at full rate, but some others might not.

That's not supported at this point.

dsix-ls2n, 2022-02-09:
Ok, fine, it would be good to put this feature in the new protocol if possible. Since the serial bandwidth will still be an issue.

closed 2025-11-14 by github-actions[bot]