PX4/PX4-Autopilot #19088: Airspeed Calibration not working on Pixhawk 5X VTOL Firmware

poyatabrizi, 2022-01-27:
**Issue Details**
When performing a calibration, the process just goes through without me having to blow into the pitot
After above-mentioned calibration, the speed with no wind reads at ~9m/s or ~30km/h
I have tested Ardupilot firmware and my standby airspeed reads around ~2m/s.

**To Reproduce**
-Connect the 45250 airspeed sensor via I2C
-Navigate to sensors and airspeed
-After sensor calibration, reboot
-Check airspeed value on the main screen

**Expected behaviour**
I expect the calibration will take into account the offset of the sensor and result in close to 0 airspeeds when there are no wind (i.e. indoor)

**Log Files and Screenshots**
Refer to forum chat where multiple people are experiencing this and there are screenshots provided.
https://discuss.px4.io/t/airspeed-sensor-is-off-by-9m-s-or-30km-h/25901

**Drone (please complete the following information):**
Pixhawk 5X flight controller
VTOL - Generic Quadplane VTOL
45250 5A1 airspeed sensor
Qground Stable Version 4.0

**Additional context**
Add any other context about the problem here.

gruffymax, 2022-01-29:
It looks as though the calibration is working as SENS_DPRES_OFF is set to the zero differential pressure value sent from the sensor. 
It seems as though this offset value is not being subtracted from the raw differential pressure before being used to calculate the airspeed.

gruffymax, 2022-01-29:
Looking through the driver for the MS4525 shows that a differential pressure is subtracted from the raw pressure in drivers/differential_pressure/ms4525/ms4525_airspeed.cpp:215
`report.differential_pressure_filtered_pa = _filter.apply(diff_press_pa_raw) - _diff_pres_offset;`

Tracing _diff_pres_offset back leads to the function Airspeed::ioctl, lib/drivers/airspeed/airspeed.cpp:114 which sets _diff_pres_offset. This function does not appear to be called anywhere so _diff_pres_offset stays as 0.0f.

ryanjAA, 2022-01-30:
Which version of px4 are you running. I’ll give this a try.

poyatabrizi, 2022-01-31:
Sorry, I forgot to mention the Firmware details. it is now added to my original comment and it is v 1.12.3

vincentpoont2, 2022-02-10:
@bkueng @dagar Any advice on this issue?

jhoexp, 2022-03-01:
Same problem here, Holybro 5x, 4525 airspeed sensor, 1.12.3, vtol standard quadplane airframe: calibration does not complete correctly (the process concludes without letting you blow into the pitot). The result is that aispeed reads large offset from 0, even when there is no wind indoor and the vehicle is standing perfectly still.

SJ-Jang, 2022-07-22:
I am currently having the same problem.
Do we have a modified version?

VTOLDavid, 2023-05-31:
Same problem here with Cube orange, Px4 1.14 beta and Sensirion sdp33

junwoo091400, 2023-06-29:
Also, this should have been fixed starting on v1.13, from this commit: https://github.com/PX4/PX4-Autopilot/commit/d1d15a6f6d46588a68458f069796213de0ba1cbf

For now I don't know if it would be valuable to backport and fix this bug in v1.12 🤷

junwoo091400, 2023-07-11:
> However, for v1.14 beta this should have been resolved, as the offset is now decoupled from the airspeed sensor driver, and is taken into account inside airspeed calibration code alone.

@VTOLDavid @SPuigUAVW could you check again whether this problem is indeed present in v1.14? It should not be here, but I don't have the setup to confirm that :cry:

VTOLDavid, 2023-07-11:
We installed v1.14 last week. We did not have the previous problem with the pitot calibration. But we had other random? problems in the setup related with the cubeorange usb drivers:
-Once flashed in last QGC version it disconnected. Only connected via USB in older versions of QGC.
-With this older version we configured a datalink but we were not able to do the pitot calibration.
-In last QGC we were able to calibrate the pitot without problem via datalink.

Then the problem seems to be the cubeorange via USB...

VTOLDavid, 2023-07-11:
We installed v1.14 last week. We had different problems with pitot calibration and USB connection:

- USB connection for cube orange does not work in latest QGC 4.2.6. Once flashed in last QGC version it disconnected. Only connected via USB in a older version of QGC 4.2.4.
-With this older version we configured a datalink.
-With QGC 4.2.6 over a telemetry link pressing pitot calibration shows nothing. The second time shows the error Command denied during calibration: 241
-With QGC 4.2.4 over a telemetry link we were able to calibrate the pitot normally.

Sems that the driver problem with cubeorange is interfering here and we had some random problems...

junwoo091400, 2023-07-18:
@ryanjAA provided this log of an Airspeed calibration working fine with the latest `release/1.14` branch, using the same SDP33 sensor: https://review.px4.io/plot_app?log=079cea16-699f-47c8-93b6-9639f5d57d00, and as you reported, the problem seems to be in QGC.

Could you create another issue on QGC repo and tag me there? @VTOLDavid

benjinne, 2023-08-14:
Hi, I'm having trouble getting the airspeed calibration menu to show up in QGC on the latest 1.14 release

I've enabled my sensor by setting SENS_EN_MS4525DO and it appears to be reading correctly when I do ms4525do status in the mavlink terminal. The readings are fairly off, so I think I do need to calibrate, however after restarting QGC the menu shown in the documentation doesn't show up:
 https://docs.px4.io/main/en/config/airspeed.html
 
 
[image]