PX4/PX4-Autopilot #19134: icm20649 glitches at low IMU_GYRO_RATEMAX on fmu-v5x on v1.12.3

simopanzer89, 2022-02-05:
**Describe the bug**
ICM-20649 IMU measurements (both accel and gyro) contain several glitches when `IMU_GYRO_RATEMAX` is set to 250 Hz on v1.12.3 (fmu-v5x).
The issue does not show up when `IMU_GYRO_RATEMAX` is set to larger values (e.g. with 400 Hz there is no issue).

**To Reproduce**
* Flash v1.12.3 stable
* From QGC reset parameters to default
* Set `IMU_GYRO_RATEMAX` to 250 Hz
* Reboot
* From the nsh console run `icm20649 status`. Several errors show up.
* Log `sensor_gyro_0` and `sensor_accel_0` for a minute. Check that glitches appear on both gyro and accel (with the autopilot fixed on ground).

```
nsh> icm20649 status
INFO  [SPI_I2C] Running on SPI Bus 3
INFO  [icm20649] FIFO empty interval: 3555 us (281.3 Hz)
icm20649: bad register: 0 events
icm20649: bad transfer: 62 events
icm20649: FIFO empty: 0 events
icm20649: FIFO overflow: 5871 events
icm20649: FIFO reset: 5872 events
icm20649: DRDY missed: 0 events
```

**Expected behavior**
The IMU measurements are expected to contain no glitches/discontinuities. It is expected that running `icm20649 status` shows no errors. The gyros should read (0,0,0) and accels should read constant (0,0,-g) when fixed on ground.

**Log Files and Screenshots**

log at `IMU_GYRO_RATEMAX` 250: glitches show up on gyros and accels
https://logs.px4.io/plot_app?log=64ffb368-d808-4687-b811-7b74288e1183

log at `IMU_GYRO_RATEMAX` 400: no glitches show up
https://logs.px4.io/plot_app?log=ae0a092a-7756-4e00-b394-129a18e8679c

Screenshots in the case of `IMU_GYRO_RATEMAX` 250: the glitches are evident even in the logfile:

[image]

[image]

**Autopilot hardware:** Holybro Pixhawk 5X (fmu-v5x)
https://docs.px4.io/master/en/flight_controller/pixhawk5x.html

IMUs:
* Accel/Gyro: ICM-20649
* Accel/Gyro: ICM-42688P
* Accel/Gyro: ICM-20602

**Additional context**
Other IMUs mounted on Pixhawk 5X (ICM-42688P and ICM-20602) are not subject to the issue.

```
nsh> icm42688p status
INFO  [SPI_I2C] Running on SPI Bus 2
INFO  [icm42688p] FIFO empty interval: 4000 us (250.0 Hz)
icm42688p: bad register: 0 events
icm42688p: bad transfer: 0 events
icm42688p: FIFO empty: 0 events
icm42688p: FIFO overflow: 0 events
icm42688p: FIFO reset: 1 events
icm42688p: DRDY missed: 0 events

nsh> icm20602 status
INFO  [SPI_I2C] Running on SPI Bus 1
INFO  [icm20602] FIFO empty interval: 4000 us (250.0 Hz)
icm20602: bad register: 0 events
icm20602: bad transfer: 0 events
icm20602: FIFO empty: 0 events
icm20602: FIFO overflow: 0 events
icm20602: FIFO reset: 1 events
icm20602: DRDY missed: 0 events
```

dagar, 2022-02-06:
Thanks @simopanzer89 I'll take a look. The scheduling for the icm20602 and icm42688p is quite nice because we can configure a FIFO watermark interrupt. For some reason the icm20649 doesn't have this feature.

Can you try with current PX4 master (soon to be v1.13)? The icm20649 scheduling was already adjusted to be a bit more flexible.

simopanzer89, 2022-02-07:
Hi @dagar , thanks for the feedback.
I tested it on the current master (052adfbfd977abfad5b6f58d5404bba7dd209736) and the issue didn't show up at `IMU_GYRO_RATEMAX` 250.
Here is the output of the status command on the various IMUs:

```
nsh> icm20649 status
INFO  [SPI_I2C] Running on SPI Bus 3
INFO  [icm20649] FIFO empty interval: 3555 us (281.3 Hz)
icm20649: bad register: 0 events
icm20649: bad transfer: 0 events
icm20649: FIFO empty: 0 events
icm20649: FIFO overflow: 0 events
icm20649: FIFO reset: 0 events
icm20649: DRDY missed: 0 events
nsh> icm42688p status
INFO  [SPI_I2C] Running on SPI Bus 2
INFO  [icm42688p] FIFO empty interval: 4000 us (250.0 Hz)
icm42688p: bad register: 0 events
icm42688p: bad transfer: 0 events
icm42688p: FIFO empty: 0 events
icm42688p: FIFO overflow: 0 events
icm42688p: FIFO reset: 0 events
icm42688p: DRDY missed: 0 events
nsh> icm20602 status
INFO  [SPI_I2C] Running on SPI Bus 1
INFO  [icm20602] FIFO empty interval: 4000 us (250.0 Hz)
icm20602: bad register: 0 events
icm20602: bad transfer: 0 events
icm20602: FIFO empty: 0 events
icm20602: FIFO overflow: 0 events
icm20602: FIFO reset: 0 events
icm20602: DRDY missed: 0 events

```

dagar, 2022-02-07:
Thanks for testing @simopanzer89, that looks better. Note the icm20649 provides 9000 Hz gyro, 4500 Hz accel, so when you request 250 Hz the actual rate is the nearest FIFO sample multiple. 

Could you let me know if you have any other issues with current master? We're closing in on a v1.13, so any feedback at this stage really helps. In the meantime if you need an icm20649 fix for v1.12 I can try to point you to the commits to cherry-pick, but we won't be doing another point release at this stage.

simopanzer89, 2022-02-08:
Thanks @dagar. I'll let you know if we have further issue with this on master.
For the time being we decided to set `IMU_GYRO_RATEMAX` to 400 Hz on v1.12.3, which actually avoids the issue.

closed 2022-02-09 by dagar