PX4/PX4-Autopilot #19146: Error in compile "make px4_sitl gazebo" in MAC OS (intel)

Desartstudio, 2022-02-08:
OS 12.1 Intel I9 
I have follow the guide in the docs for use gazebo with px4 in my mac book
I have intalled Gazebo e it work fine
I compile px4_sitl fine

When i try to "make px4_sitl gazebo" i receive some error i have search and i think is gstreammer the problem 

[This issue](https://github.com/PX4/PX4-Autopilot/issues/13416) 
Propose to fix in utbuntu with  this commands 

```
sudo apt install libgstreamer1.0-dev
sudo apt install gstreamer1.0-plugins-good
sudo apt install gstreamer1.0-plugins-bad
sudo apt install gstreamer1.0-plugins-ugly
```

I search in brew this package .. and have it 

`brew list | grep gs       
gst-libav
gst-plugins-bad
gst-plugins-base
gst-plugins-good
gst-plugins-ugly
gstreamer
`
The only one not are is libgstreamer1.0-dev but in mac not exist the package

This is the complete output :

`utente@MBP-di-Antonio PX4-Autopilot % make px4_sitl gazebo
[0/11] git submodule Tools/sitl_gazebo
[6/11] Performing configure step for 'sitl_gazebo'
-- install-prefix: /usr/local
-- cmake build type: RelWithDebInfo
-- The C compiler identification is AppleClang 13.0.0.13000029
-- The CXX compiler identification is AppleClang 13.0.0.13000029
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Performing Test COMPILER_SUPPORTS_CXX17
-- Performing Test COMPILER_SUPPORTS_CXX17 - Success
-- Performing Test COMPILER_SUPPORTS_CXX14
-- Performing Test COMPILER_SUPPORTS_CXX14 - Success
-- Performing Test COMPILER_SUPPORTS_CXX11
-- Performing Test COMPILER_SUPPORTS_CXX11 - Success
-- Performing Test COMPILER_SUPPORTS_CXX0X
-- Performing Test COMPILER_SUPPORTS_CXX0X - Success
-- Using C++17 standard
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Found Boost: /usr/local/lib/cmake/Boost-1.76.0/BoostConfig.cmake (found suitable version "1.76.0", minimum required is "1.58") found components: system thread filesystem 
CMake Warning (dev) at /usr/local/Cellar/cmake/3.22.2/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (PkgConfig)
  does not match the name of the calling package (gazebo).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  /usr/local/Cellar/cmake/3.22.2/share/cmake/Modules/FindPkgConfig.cmake:99 (find_package_handle_standard_args)
  /usr/local/lib/cmake/gazebo/gazebo-config.cmake:62 (include)
  CMakeLists.txt:48 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

[... 221 lines cut ...]
/Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include/moc_gazebo_user_camera_plugin.cpp:29:5: note: expanded from macro 'QT_MOC_LITERAL'
    Q_STATIC_BYTE_ARRAY_DATA_HEADER_INITIALIZER_WITH_OFFSET(len, \
    ^
/Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include/moc_gazebo_user_camera_plugin.cpp:66:1: error: unknown type name 'QT_INIT_METAOBJECT'
QT_INIT_METAOBJECT const QMetaObject gazebo::UserCameraPlugin::staticMetaObject = { {
^
/Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo/include/moc_gazebo_user_camera_plugin.cpp:66:37: error: expected ';' after top level declarator
QT_INIT_METAOBJECT const QMetaObject gazebo::UserCameraPlugin::staticMetaObject = { {
                                    ^
                                    ;
7 errors generated.
[128/130] Building CXX object CMakeFiles/gazebo_user_camera_plugin.dir/src/gazebo_user_camera_plugin.cpp.o
FAILED: CMakeFiles/gazebo_user_camera_plugin.dir/src/gazebo_user_camera_plugin.cpp.o 
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/c++ -DBOOST_ALL_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_REGEX_DYN_LINK -DBOOST_SYSTEM_DYN_LINK -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DLIBBULLET_VERSION=3.20 -DLIBBULLET_VERSION_GT_282 -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_DEBUG -DQT_WIDGETS_LIB -Dgazebo_user_camera_plugin_EXPORTS -I/Users/utente/Documents/GIT/PX4/PX4-Autopilot/Tools/sitl_gazebo/include -I/Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo -I/usr/local/include/eigen3/eigen3 -I/usr/local/include/gazebo-11/gazebo/msgs -I/Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/mavlink -I/Users/utente/Documents/GIT/PX4/PX4-Autopilot/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/local/Cellar/libffi/3.4.2/include -I/usr/local/Cellar/gstreamer/1.18.5/include/gstreamer-1.0 -I/usr/local/Cellar/glib/2.70.3/include -I/usr/local/Cellar/glib/2.70.3/include/glib-2.0 -I/usr/local/Cellar/glib/2.70.3/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.45/include -I/usr/local/Cellar/gst-plugins-base/1.18.5/include/gstreamer-1.0 -isystem /usr/local/include -isystem /usr/local/include/gazebo-11 -isystem /usr/local/Cellar/bullet/3.21/include/bullet -isystem /usr/local/Cellar/bullet/3.21/include -isystem /usr/local/include/simbody -isystem /usr/local/Cellar/sdformat9/9.7.0/include/sdformat-9.7 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Paging -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /usr/local/opt/qt@5/lib -isystem /usr/local/opt/qt@5/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt@5/./mkspecs/macx-clang -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/opencv/4.5.4_4/include/opencv4 -isystem /usr/local/Cellar/fcl/0.7.0/include -isystem /usr/local/Cellar/assimp/5.2.1/include -isystem /usr/local/include/sdformat-9.7/sdf/.. -isystem /usr/local/include/ignition/cmake2 -isystem /usr/local/opt/qt@5/lib/QtWidgets.framework/Headers -isystem /usr/local/opt/qt@5/lib/QtGui.framework/Headers -isystem /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk/System/Library/Frameworks/OpenGL.framework/Headers -O2 -g -DNDEBUG -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk -fPIC -F/Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk/System/Library/Frameworks  -Wno-deprecated-declarations -Wno-address-of-packed-member -fPIC -std=gnu++17 -MD -MT CMakeFiles/gazebo_user_camera_plugin.dir/src/gazebo_user_camera_plugin.cpp.o -MF CMakeFiles/gazebo_user_camera_plugin.dir/src/gazebo_user_camera_plugin.cpp.o.d -o CMakeFiles/gazebo_user_camera_plugin.dir/src/gazebo_user_camera_plugin.cpp.o -c /Users/utente/Documents/GIT/PX4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_user_camera_plugin.cpp
In file included from /Users/utente/Documents/GIT/PX4/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_user_camera_plugin.cpp:46:
In file included from /Users/utente/Documents/GIT/PX4/PX4-Autopilot/Tools/sitl_gazebo/include/gazebo_user_camera_plugin.h:46:
/usr/local/include/gazebo-11/gazebo/gui/GuiPlugin.hh:46:11: error: no member named 'init' in 'QStyleOption'
      opt.init(this);
      ~~~ ^
1 error generated.
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build /Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.22.2/bin/cmake --build /Users/utente/Documents/GIT/PX4/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 16
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1`

Desartstudio, 2022-02-16:
No reply?

Desartstudio, 2022-02-17:
I have resolved, 
using brew uninstall QT and dependence .. i think are a conflict to qt@5

closed 2022-02-17 by Desartstudio

kevinuav, 2023-11-02:
Hi, I encounter the same problem. And I fixed it after:"brew uninstall --ignore-dependencies QT". Thank you. But I just don't understand why the gazebo still can work without the QT dependence? Is not the QT a necessary component which is needed by gazebo? If it is false. Then why the installation process installs a needless component?