PX4/PX4-Autopilot #19150: Airspeed driver not started on boot w/ 1.13 master

ryanjAA, 2022-02-08:
Using master from today (86f8168) on pixhawk 4 airspeed driver is not started automatically. 

I get no airspeed found when trying to do an airspeed cal. If I start the driver manually I can calibrate it, etc.

Tried with an ms4525

dagar, 2022-02-08:
Did you set the `SENS_EN_MS4525` parameter?

ryanjAA, 2022-02-09:
No - I didn't even know that was there (but actually makes sense since there are so many airspeeds). Closing. this.

closed 2022-02-09 by ryanjAA

dagar, 2022-02-09:
We had to move away from automatically starting by default because there was a growing number of potential conflicts with sensors that don't have any real way to uniquely identify. Initially it's a slight inconvenience (especially if you don't know the parameter exists), but I think ultimately explicit is better.

ryanjAA, 2022-02-09:
Totally agreed, makes more sense just will take some getting used to. Thanks