PX4/PX4-Autopilot #19169: Rover doesn't stop in offboard mode

JohnnyLiu2886, 2022-02-11:
According to RoverPositionControl, rover should stop when the distance to waypoint is smaller than the loiter radius. It sets INDEX_YAW and INDEX_THROTTLE to 0.0f and publish it through uorb messages actuator_controls, vehicle_torque_setpoint and vehicle_thrust_setpoint.
[image]

However, when the distance to waypoint is smaller than 2 (the default NAV_LOITER_RAD), all the PWM becomes 1100.
In my setting, the minimum PWM is 1100. It means that the PWM outputs are set to the minimum when throttle is 0.
[image]

However, I think it shouldn't be happened. The PWM for the rover to stop should not be the minimum PWM. It should be the middle one as minimum PWM means reverse motion for the motors. It means negative throttle.

As you can see here, when I disarm the rover, the PWM outputs are 1500. 1500 is the real PWM for stopping the rover.
[image]

Apart from that, I am using aion Robotics R1 UGV as the airframe of the rover. 
I also find the following:
''''''''''''''''''''''''''
# TODO: Set to -1.0, to allow reversing. This will require many changes in the codebase
# to support negative throttle.
param set-default GND_THR_MIN 0
''''''''''''''''''''''''''

Does it mean when throttle is 0, the pwm will be minimum in default?

Does anyone know how to set the PWM outputs to 1500 when trajectory_setpoint reaches its destination in offboard mode?
My PWM range is from 1100 to 1900.

There is no problem for the RC manual control mode. The problem only happens for offboard control

junwoo091400, 2022-12-02:
This should now be possible as https://github.com/PX4/PX4-Autopilot/pull/15949 got merged!

chfriedrich98, 2025-02-11:
Closing this issue since it references the outdated rover module.
If this problem is present in the updated version, please open another issue.

closed 2025-02-11 by chfriedrich98