PX4/PX4-Autopilot #19176: IC Engine RPM data over telemetry?

dystopic1, 2022-02-11:
Hi everyone. I want to see rpm value of an internal combustion engine in QGC? Is there a stand alone feature in PX4 or we need a special hardware?

dagar, 2022-02-11:
There are various pieces in PX4 for logging/displaying/etc internal combustion engine data or data from an RPM sensor, but the first thing to figure out is how you're getting that data in into the system. https://github.com/PX4/PX4-Autopilot/blob/master/msg/internal_combustion_engine_status.msg

What do you have exactly? Does it have any kind of serial or CAN interface?

roman-dvorak, 2023-07-31:
Hi @dystopic1,

You can display the RPM value of an internal combustion engine in QGroundControl using our [TFRPM](https://docs.px4.io/main/en/sensor/thunderfly_tachometer.html#thunderfly-tfrpm01-revolution-counter) sensor. If the engine already has an integrated RPM sensor (with TTL levels), you can simply connect it to the TFRPM sensor. However, if the engine doesn't have a built-in RPM sensor, you'll need to add one probe. You can choose either an optical sensor with a coding disk or a Hall-effect sensor with magnets on the rotational axis.

In PX4, we have a TFRPM driver that publishs uORB RPM messages, and these messages are sent through MAVLink to QGroundControl. There, you can visualize the RPM values in the table displaying the drone's parameters, as shown in the screenshot.

[image]

RPM data are continuously logged to internal ulog file. 

More informations about RPM sensor is here: https://github.com/ThunderFly-aerospace/TFRPM01