PX4/PX4-Autopilot #19185: SITL run incorrectly,probably wrong config for gzserver?

haozi6, 2022-02-13:
hello,
    I have using SITL for months,I'm sure it's work correctly about 20 days ago.
   When I "make px4_sitl gazebo" or "make px4_sitl gazebo_boat",it does not work,I try to solve it few days. 
   
   ```shell
   .......
   [Msg] Waiting for master.
   [Err] [ConnectionManager.cc:110] Failed to connect to master in 30 seconds.
   [Err] [gazebo_shared.cc:78] Unable to initialize transport.
   [Err] [gazebo_client.cc:56] Unable to setup Gazebo
   ```
   gazebo verion:11.90    
   I also try the verision 9.x.x
   ubuntu20.04

   now I have no idea what to do,could you help me?

Jaeyoung-Lim, 2022-02-13:
@haozi6 Can you run gazebo without PX4?

eastman-bechamo, 2022-06-16:
I'm having the same issue on Ubuntu 20.04. I'm able to run gazebo without PX4.