PX4/PX4-Autopilot #19188: PWM outputs go high for ~500ms on boot

spiderkeys, 2022-02-14:
I noticed today that various peripherals on the vehicle I am doing bringup on were actuating upon power-on, for about 527ms.

Most well-behaved servo devices ignore this, because it is not modulating at an expected 50-400Hz rate, but other devices, such as lights and other PWM->Analog Output type peripherals are seeing this as a "turn on at 100%" signal.

Is there a known way to prevent this behavior?

Screenshot below of an aux PWM output and a motor output.

[image]

dagar, 2022-02-14:
@davids5

hhenryy, 2022-02-14:
My case is not necessary related to the above, but I have seen on powering down (but I think also on powering up) that our servo's will wiggle for a tiny bit. I haven't investigated this too much because it happens seldom that the servo's actually actuate.

spiderkeys, 2022-02-14:
@davids5 changing that did indeed keep the PWM throughout startup. Thanks!

ryanjAA, 2022-03-03:
I’ve seen this too fyi. Especially on reboot.

spiderkeys, 2022-04-16:
@davids5 we noticed recently during a firmware upgrade that this may also be happening while running the bootloader.

I'd like to try and apply the same approach as above to see if it fixes things, but `io_timer_channel_get_gpio_output()` isn't accessible in the bootloader project. Of course, I can manually set the GPIO, but I was wondering if you had ideas for how to approach this in a more generic manner.

spiderkeys, 2022-04-16:
So I captured a trace of the full process, from DFU mode, to reset, to bootloader, through flashing PX4 firmware, and then starting the PX4 app. What I found was that one of the servo pins (and possibly others I don't yet know about) is going high when put into DFU mode, otherwise it appears everything that everything is low through the bootloader process until explicit activation of the pins via PX4.

I found this post, which sounds a lot like what I am seeing:
https://community.st.com/s/question/0D53W00000TpRWFSA3/pin-state-of-stm32g431kbt-in-dfu-boot-mode

Not sure there is really a way to work around this one on this off the shelf FMU (Matek H743 Wing v1), so I guess the only resolution in this case is to make sure power is disconnected to the rest of the system whenever flashing the bootloader. Kind of problematic if we wanted to do a bootloader update in the field, but otherwise manageable.

Let this serve as a warning to others I guess...

spiderkeys, 2022-04-16:
Given the above behavior on this FMU, it leaves me feeling a bit nervous about what state the GPIO config is in during the bootloader, so I also took the following steps in `board_on_reset()` in bootloader_main.c to get the same behavior as the main firmware:

```c
#include <px4_arch/io_timer.h>
#include <px4_platform_common/init.h>
#include <px4_platform/gpio.h>

extern int sercon_main(int c, char **argv);

static void init_timer_pin_gpio_out_low( uint32_t timer_base, uint32_t port_pin )
{
	uint32_t gpio_af = 0;

	switch(timer_base) {
		case STM32_TIM1_BASE:
			gpio_af = GPIO_AF1;
			break;

		case STM32_TIM4_BASE:
		case STM32_TIM5_BASE:
			gpio_af = GPIO_AF2;
			break;

		case STM32_TIM8_BASE:
			gpio_af = GPIO_AF3;
			break;

		case STM32_TIM15_BASE:
			gpio_af = GPIO_AF4;
			break;

		default:
			return;
	}

	// uint32_t gpio_in = gpio_af | (GPIO_ALT | GPIO_SPEED_50MHz | GPIO_FLOAT) | port_pin;
	uint32_t gpio_out = gpio_af | (GPIO_ALT | GPIO_SPEED_50MHz | GPIO_PUSHPULL) | port_pin;

	px4_arch_configgpio( gpio_out );
}

static uint32_t timer_base[ DIRECT_PWM_OUTPUT_CHANNELS] = {
	STM32_TIM5_BASE,
	STM32_TIM5_BASE,
	STM32_TIM5_BASE,
	STM32_TIM5_BASE,
	STM32_TIM4_BASE,
	STM32_TIM4_BASE,
	STM32_TIM4_BASE,
	STM32_TIM4_BASE,
	STM32_TIM8_BASE,
	STM32_TIM15_BASE,
	STM32_TIM15_BASE,
	STM32_TIM1_BASE
};

static uint32_t pwm_pins[DIRECT_PWM_OUTPUT_CHANNELS] = {
	(GPIO_PORTA | GPIO_PIN0),
	(GPIO_PORTA | GPIO_PIN1),
	(GPIO_PORTA | GPIO_PIN2),
	(GPIO_PORTA | GPIO_PIN3),
	(GPIO_PORTD | GPIO_PIN12),
	(GPIO_PORTD | GPIO_PIN13),
	(GPIO_PORTD | GPIO_PIN14),
	(GPIO_PORTD | GPIO_PIN15),
	(GPIO_PORTB | GPIO_PIN1),
	(GPIO_PORTE | GPIO_PIN5),
	(GPIO_PORTE | GPIO_PIN6),
	(GPIO_PORTA | GPIO_PIN8)
};

__EXPORT void board_on_reset(int status)
{
	// Configure the GPIO pins to outputs and keep them low.
	for (int i = 0; i < DIRECT_PWM_OUTPUT_CHANNELS; ++i) {
		init_timer_pin_gpio_out_low( timer_base[i], pwm_pins[i] );
	}
}
```

dagar, 2022-04-16:
For the in tree bootloaders (H7 boards and CAN nodes) we could start unifying more of the init logic between bootloader and app.

davids5, 2022-04-27:
@spiderkeys - see https://github.com/PX4/PX4-Autopilot/pull/19535 it can be replicated on all H7.