PX4/PX4-Autopilot #19190: actuator output function, how to use it ?

dookei, 2022-02-14:
Hi I would like to write a small application to control some servo motors.
I saw that this PR [https://github.com/PX4/PX4-Autopilot/pull/18265](https://github.com/PX4/PX4-Autopilot/pull/18265) should support that already. 
But I still cant find a way to do what I need. Should I publish to a topic in any form ? 
Any guidance would be appreciated.

bkueng, 2022-02-15:
Did you configure the system according to https://docs.px4.io/master/en/config/actuators.html? Then you can set an output to `Offboard_Actuator_Setx` and use [MAV_CMD_DO_SET_ACTUATOR](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR) via mavlink.

dookei, 2022-02-16:
Hi @bkueng thank you for you reply. I will try that now. 
Is there any chance that  the outputs can be set via uorb publish ? I would like to keep and contain  the control from the servomotors only within the FC without giving the option to control it over mavlink. "Closed loop" kind?
My idea is to avoid miss usage from some servo motors. 

**For example:** 
I have a task that is checking several Digital and Analogue inputs plus PX4 status(armed?, landed?, height, position etc). The task with this information then decides the angle, the speed and other parameters from the servos.
Thanks !

dookei, 2022-02-16:
I been trying to control the servos within px4 ...but without success.
According to the mavlink [MAV_CMD_DO_SET_ACTUATOR](https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_ACTUATOR) this is going to publish on the actuator_controls_0 topic.
https://github.com/PX4/PX4-Autopilot/blob/abfa3d23a5fd7d0bd31214882cbddd3b75943348/src/modules/mavlink/mavlink_receiver.cpp#L555-L561

When I try to publish on this topic i get no reaction at all.
Is there a way to publish to mavlink inside Px4?

bkueng, 2022-02-17:
>  I would like to keep and contain the control from the servomotors only within the FC without giving the option to control it over mavlink. "Closed loop" kind?

For pure fmu internal use, I suggest you add your own uorb topic with new output functions (https://github.com/PX4/PX4-Autopilot/blob/master/src/lib/mixer_module/output_functions.yaml) and then handle them here: https://github.com/PX4/PX4-Autopilot/blob/master/src/lib/mixer_module/functions.hpp#L85.

dookei, 2022-02-17:
Many thanks for your help. means alot to me. 

Just so You understand my objective. I have a total of 6 servos. 2 for Lock mechanism, and 4 for release and pick the cargo.
I need to control each single individually and with different values. For example servo 1 from lock is reversed and needs different operation range from the other Lock servo. The same operation principle applies to the elevator servos. 
They should only open when disarmed and landed. And close armed and disarmed...plus some other conditions that need to be met.  
[image]
[image]
As you proposed I am adding the required functions for  the cargo operation  (for now just the cargo_lock)
https://github.com/PX4/PX4-Autopilot/blob/master/src/lib/mixer_module/output_functions.yaml
```c++
    # Add your own here:
    #MyCustomFunction: 10000
    # Cargo
    Cargo_Lock:
      start: 10000
      count: 2
    Cago_Elevator:
      start: 10010
      count: 4
```
https://github.com/PX4/PX4-Autopilot/blob/ca97b9ba5f1754129177b853963df022bdebc271/src/lib/mixer_module/functions.hpp#L277
```c++
/**
 * Functions: Cargo_Lock
 */
class FunctionCargoLock : public FunctionProviderBase
{
public:
	FunctionCargoLock();
	static FunctionProviderBase *allocate(const Context &context) { return new FunctionCargoLock(); }

	void update() override;
	float value(OutputFunction func) override { return _data[(int)func - (int)OutputFunction::Cargo_Lock1]; }

private:
	static constexpr int max_num_locks =2;
	uORB::Subscription _topic{ORB_ID(cargo_lock)};
	float _data[max_num_locks];
};

/**
 * Functions: Cargo_Elevator
 */
class FunctionCargoElevator : public FunctionProviderBase
{
public:
	FunctionCargoElevator() = default;
	static FunctionProviderBase *allocate(const Context &context) { return new FunctionCargoElevator(); }

	void update() override;
	float value(OutputFunction func) override { return _data[(int)func - (int)OutputFunction::CargoElevator_Set1]; }

private:
	static constexpr int max_num_elevators = 4;
	uORB::Subscription _topic{ORB_ID(cargo_elevator)};
	float _data[max_num_elevators];
};
```
I create a new message under https://github.com/PX4/PX4-Autopilot/tree/master/msg

```
uint64 timestamp			# time since system start (microseconds)
uint64 timestamp_sample	    # the timestamp the data this control response is based on was sampled
uint8 INDEX_LOCK_LEFT = 0
uint8 INDEX_LOCK_RIGHT = 1

uint8 NUM_LOCKS = 2
float32[2] control # range: [-1, 1], where 1 means maximum positive position,
                   # -1 maximum negative,
                   # and NaN maps to disarmed
```
Add it to CMake 
https://github.com/PX4/PX4-Autopilot/blob/b0f9611eb9da61eefceadb4b10933c92a5770755/msg/CMakeLists.txt#L57
`	cargo_lock.msg
`
After  https://github.com/PX4/PX4-Autopilot/blob/b0f9611eb9da61eefceadb4b10933c92a5770755/src/lib/mixer_module/functions.cpp#L126-L135
I had my two functions
```c++
FunctionCargoLock::FunctionCargoLock()
{
	for (int i = 0; i < max_num_locks; ++i) {
		_data[i] = NAN;
	}
}

void FunctionCargoLock::update()
{
	cargo_lock_s cargo_lock;

	if (_topic.update(&cargo_lock)) {
		_data[0] = cargo_lock.control[0];
		_data[1] = cargo_lock.control[1];
	}
}

```

after https://github.com/PX4/PX4-Autopilot/blob/c13726af66e73d5664d158e8824ff8287e4f1b70/src/lib/mixer_module/mixer_module.cpp#L66
added 
	{OutputFunction::Cargo_Lock1, OutputFunction::Cargo_Lock2, &FunctionCargoLock::allocate},

**Quick Note:** 
I realize that I am pretty much copy pasting the same as the actuators_set or gimbal.
Every thing compiles fine. 

over nsh i change the PWM_MAIN_FUNC to my new function 
```
nsh> param show PWM_MAIN_FUNC*
Symbols: x = used, + = saved, * = unsaved
x + PWM_MAIN_FUNC1 [674,1198] : 10000
x + PWM_MAIN_FUNC2 [675,1199] : 10001

```
This can be seen under QGroundControl as well and the servos can also be tested.
[image]

Create a short test application that publishs to the cargo_lock topic
```
void test (void){

	uint64_t timestamp_us = hrt_absolute_time();

	struct cargo_lock_s cargo_lock;
	memset(&cargo_lock, 0, sizeof(cargo_lock));

  cargo_lock_pub = orb_advertise(ORB_ID(cargo_lock), &cargo_lock);

	cargo_lock.timestamp =timestamp_us;
  cargo_lock.control[0]=1;
  cargo_lock.control[1]=1;
	orb_publish(ORB_ID(cargo_lock), cargo_lock_pub, &cargo_lock);

  sleep(2);
  timestamp_us = hrt_absolute_time();
	cargo_lock.timestamp =timestamp_us;
  cargo_lock.control[0]=-1;
  cargo_lock.control[1]=-1;
	orb_publish(ORB_ID(cargo_lock), cargo_lock_pub, &cargo_lock);
}
int cargo_test_main(int argc, char *argv[])
{
  PX4_INFO("START");
  test();
  PX4_INFO("DONE and out");
	return 0;
}
```
Which also works  fine, I can see that after running the topic was published with data.
```
nsh> listener cargo_lock

TOPIC: cargo_lock
 cargo_lock
    timestamp: 28724471 (162.714249 seconds ago)
    timestamp_sample: 0
    control: [-1.0000, -1.0000]
```

But this doesn't seem to be subscribed by the mixer_module. Is there anything that I am missing ?

Sorry for the recurring questions. I am just trying to understand the best and simplest way to implement something that can be used in general.
 Thanks once again!

bkueng, 2022-02-18:
That looks correct. Are the test sliders moving the servo? Can you check the output of `px4io status`?

dookei, 2022-02-18:
Yes the sliders can move the servo.
The output from px4io status also looks good. The topic from cargo_lock looks also good..
I am getting out of ideas what might be missing. Any thoughts?
Can be that the mixer is not subscribing properly to the topic?  Any method I could debug that?
```
nsh> px4io status
protocol 5 hardware 2 bootloader 3 buffer 64B crc 0x1388160798
8 controls 8 actuators 18 R/C inputs 2 analog inputs
 px4io_status
    timestamp: 116403185 (0.058818 seconds ago)
    voltage_v: 5.3940
    rssi_v: 1.6620
    free_memory_bytes: 1848
    pwm: [0, 0, 900, 900, 900, 900, 900, 900]
    pwm_disarmed: [0, 0, 900, 900, 900, 900, 900, 900]
    pwm_failsafe: [0, 0, 900, 900, 900, 900, 900, 900]
    pwm_rate_hz: [400, 400, 400, 0, 0, 0, 0, 0]
    raw_inputs: [0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
    status_arm_sync: True
    status_failsafe: False
    status_fmu_initialized: True
    status_fmu_ok: True
    status_init_ok: True
    status_outputs_armed: True
    status_raw_pwm: True
    status_rc_ok: False
    status_rc_dsm: False
    status_rc_ppm: False
    status_rc_sbus: False
    status_rc_st24: False
    status_rc_sumd: False
    status_safety_off: True
    alarm_pwm_error: False
    alarm_rc_lost: True
    arming_failsafe_custom: True
    arming_fmu_armed: False
    arming_fmu_prearmed: False
    arming_force_failsafe: False
    arming_io_arm_ok: True
    arming_lockdown: False
    arming_termination_failsafe: False

0 raw R/C inputs
ADC inputs 2225 2109
features 0x0004 RSSI_ADC
sbus rate 72
debuglevel 0

IMU heater off
INFO  [mixer_module] Param prefix: PWM_MAIN
control latency: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms
Channel Configuration:
Channel 0: func: 10000, value: 0, failsafe: 65535, disarmed: 0, min: 600, max: 2200
Channel 1: func: 10001, value: 0, failsafe: 65535, disarmed: 0, min: 600, max: 2200
Channel 2: func: 0, value: 900, failsafe: 65535, disarmed: 900, min: 1000, max: 2000
Channel 3: func: 0, value: 900, failsafe: 65535, disarmed: 900, min: 1000, max: 2000
Channel 4: func: 0, value: 900, failsafe: 65535, disarmed: 900, min: 1000, max: 2000
Channel 5: func: 0, value: 900, failsafe: 65535, disarmed: 900, min: 1000, max: 2000
Channel 6: func: 0, value: 900, failsafe: 65535, disarmed: 900, min: 1000, max: 2000
Channel 7: func: 0, value: 900, failsafe: 65535, disarmed: 900, min: 1000, max: 2000

```

Output from uorb status also show that the topic car_lock is created and published properly 
```
nsh> uorb status
TOPIC NAME                     INST #SUB #Q SIZE PATH
actuator_armed                    0   15  1   16 /obj/actuator_armed0
actuator_controls_0               0    7  1   48 /obj/actuator_controls_00
actuator_controls_status_0        0    0  1   24 /obj/actuator_controls_status_00
actuator_motors                   0    2  1   56 /obj/actuator_motors0
actuator_outputs                  0    3  1   80 /obj/actuator_outputs0
actuator_outputs                  1    2  1   80 /obj/actuator_outputs1
actuator_outputs                  2    1  1   80 /obj/actuator_outputs2
actuator_outputs                  3    0  1   80 /obj/actuator_outputs3
actuator_servos_trim              0    0  1   40 /obj/actuator_servos_trim0
actuator_test                     0    4  8   24 /obj/actuator_test0
adc_report                        0    1  1   96 /obj/adc_report0
autotune_attitude_control_status  0    3  1  104 /obj/autotune_attitude_control_status0
battery_status                    0    8  1  168 /obj/battery_status0
cargo_lock                        0    1  1   24 /obj/cargo_lock0

```

bkueng, 2022-02-18:
Could you try with the AUX pins? The IO possibly only starts to update while armed (depending on configuration and safety switch).

dookei, 2022-02-18:
Ok on the AUX output it works. That's some progress :+1:
Although I can only use the servo when armed. Disarmed it doesn't react.
You have any idea why it doesn't work on the MAIN? I'm actually using/need the AUX for the DSHOT ESCs.

spiderkeys, 2022-02-19:
I also ran into the same issue, regarding Dshot outputs not being active when disarmed, even when going through `actuator_test`.

Ultimately, this seems linked to the fact that `control_allocator` is not publishing on the `actuator_motors` topic when disarmed, as `do_update` within the dynamic mixer update will be false due to lack of torque/thrust commands, bypassing the call to `publish_actuator_controls()`. The information published on this topic is important to the actuator test in at least two ways that I can see.

1. It contains the information about the motor being reversible. Without this being published, the `FunctionMotors` class never gets a copy of `actuator_motor_s` data, which contains the reversible flags that it acts on, so all of its `updateValues` calls will be passing in 0 for the `reversible` argument when driven by the `actuator_test` command. The output seems to be fine if the system is actually armed and receiving control setpoints, as this triggers publications by `control_allocator`.

2. For Dshot specifically, the dshot commands never get triggered. This is because within `mixing_module`'s dynamic mixer update, the final call to _interface.updateOutputs() gets passed in `has_updates == false`. The `has_updates` argument is determined in `MixingOutput::updateDynamicMixer` based on:
```c++
bool has_updates = _subscription_callback && _subscription_callback->updated();
```
This is always false, because the subscription is the same `actuator_motors` subscription within `FunctionMotors` that depends on `control_allocator` publishing those values.

All of that said, I'm not sure what the most appropriate fix is here, since both of these problems need to be solved (at least for 3D Dshot to function correctly). My first guess is that publishing `actuator_motors` from `control_allocator` regardless of whether or not torque/thrust setpoints are received would be one way to fix the issue. I don't have a deep enough understanding yet to know whether or not that would impact other PWM/UAVCAN/etc functions.

dookei, 2022-02-21:
Update:
I was confused by a small short noise(click) made by the servo on MAIN1. So i measured the output with a logic analyzer.
The signal was there....just the servo didn't seem to react. Since my application was only setting to minimum(800) and maximum(2200) it had to be something with the range that was not being accepted.. I believe the servos I have only operate in the 1000 and 2000 ms region. I change the operation range and now they work as the should. So that problem is solved. 

**The second issue** that I am facing is the same as the @spiderkeys with the **DSHOT**(PWM seems to work).I can confirm the behavior described.

[image]
DISARMED state
[image]
[image]

Once I arm it, nothing is output.

bkueng, 2022-02-22:
@dookei can you provide the output of `uorb top` `param show -c`, `dshot status` and `control_allocator status`? I'm interested to know why exactly there's no updates after arming, as I've tested with pretty much the same configuration from what I can tell.

dookei, 2022-02-23:
@bkueng 
here is the output from 
`uorb top param show -c`
```
update: 1s, topics: 106, total publications: 0, 0.0 kB/s
TOPIC NAME     INST #SUB RATE #Q SIZE
parameter_update  0   41    0  1   40
```
and from `control_allocator status`

```
nsh> control_allocator status
INFO  [control_allocator] Running
INFO  [control_allocator] Method: Auto
INFO  [control_allocator] Effectiveness Source: Multirotor
INFO  [control_allocator]   Effectiveness.T =
	-0.97500002	1.1700001	0.32500002	       0	       0	    -6.5
	0.97500002	-1.1700001	0.32500002	       0	       0	    -6.5
	0.97500002	1.1700001	-0.32500002	       0	       0	    -6.5
	-0.97500002	-1.1700001	-0.32500002	       0	       0	    -6.5
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0
	       0	       0	       0	       0	       0	       0

INFO  [control_allocator]   minimum =
	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0

INFO  [control_allocator]   maximum =
	       1	       1	       1	       1	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0	       0

INFO  [control_allocator]   Configured actuators: 4
control_allocator: cycle: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms

```

bkueng, 2022-02-23:
`uorb top` and `param show -c` are separate commands.

`control_allocator: cycle: 0 events, 0us elapsed, 0.00us avg, min 0us max 0us 0.000us rms`
Something isn't running correctly, do you have other code modifications?

dookei, 2022-02-23:
Your right my bad.
update: 1s, topics: 106, total publications: 8714, 905.8 kB/s
TOPIC NAME                     INST #SUB RATE #Q SIZE
actuator_armed                    0   11    2  1   16 
actuator_outputs                  0    3   20  1   80 
actuator_outputs                  2    1   20  1   80 
adc_report                        0    1  100  1   96 
battery_status                    0    7  100  1  168 
commander_state                   0    1    2  1   16 
cpuload                           0    4    2  1   16 
estimator_attitude                0    2  200  1   56 
estimator_attitude                1    1  200  1   56 
estimator_innovation_test_ratios  0    1   99  1  128 
estimator_innovation_test_ratios  1    1  100  1  128 
estimator_innovation_variances    0    1   99  1  128 
estimator_innovation_variances    1    1  100  1  128 
estimator_innovations             0    1   99  1  128 
estimator_innovations             1    1  100  1  128 
estimator_local_position          0    2   99  1  168 
estimator_local_position          1    1  100  1  168 
estimator_odometry                0    1   99  1  256 
estimator_selector_status         0    7    1  1  160 
estimator_sensor_bias             0    4    1  1  120 
estimator_sensor_bias             1    1    1  1  120 
estimator_states                  0    1   99  1  216 
estimator_states                  1    1  100  1  216 
estimator_status                  0    5   99  1  112 
estimator_status                  1    2  100  1  112 
estimator_status_flags            0    1    1  1   88 
estimator_status_flags            1    1    1  1   88 
failure_detector_status           0    1    2  1   24 
magnetometer_bias_estimate        0    2   47  1   64 
px4io_status                      0    1    1  1  144 
rtl_time_estimate                 0    2    1  1   24 
safety                            0    2    1  1   16 
sensor_accel                      0    6  400  8   48 
sensor_accel                      1    4  406  8   48 
sensor_baro                       0    2   75  1   32 
sensor_combined                   0    3  200  1   48 
sensor_gyro                       0    6  400  8   48 
sensor_gyro                       1    4  401  8   48 
sensor_gyro_fft                   0    1   82  1  104 
sensor_gyro_fifo                  1    2  400  4  224 
sensor_mag                        0    5   47  4   40 
sensors_status_imu                0    2  200  1   96 
system_power                      0    2  100  1   40 
telemetry_status                  0    2    1  1   88 
telemetry_status                  1    2    1  1   88 
vehicle_acceleration              0    2  200  1   32 
vehicle_air_data                  0   10   19  1   40 
vehicle_angular_acceleration      0    2  398  1   32 
vehicle_angular_velocity          0    9  398  1   32 
vehicle_attitude                  0    7  200  1   56 
vehicle_attitude_setpoint         0    4  200  1   64 
vehicle_control_mode              0   15    2  1   24 
vehicle_imu                       0    5  200  1   56 
vehicle_imu                       1    5  200  1   56 
vehicle_imu_status                0    7   10  1  120 
vehicle_imu_status                1    4   10  1  120 
vehicle_land_detected             0   12    1  1   24 
vehicle_local_position            0   15   99  1  168 
vehicle_magnetometer              0    4   11  1   40 
vehicle_odometry                  0    1   99  1  256 
vehicle_rates_setpoint            0    4  200  1   32 
vehicle_status                    0   26    2  1   88 
vehicle_status_flags              0    5    2  1   48 
`param show -c 
`
```

[... 92 lines cut ...]
x + RC2_MAX [755,1317] : 2006.0000
x + RC2_MIN [756,1318] : 982.0000
x + RC2_TRIM [758,1320] : 1494.0000
x + RC3_MAX [760,1322] : 2006.0000
x + RC3_MIN [761,1323] : 982.0000
x + RC3_TRIM [763,1325] : 1491.0000
x + RC4_MAX [765,1327] : 2006.0000
x + RC4_MIN [766,1328] : 982.0000
x + RC4_TRIM [768,1330] : 1493.0000
x + RC_CHAN_CNT [795,1357] : 18
x + RC_MAP_ARM_SW [801,1363] : 8
x + RC_MAP_AUX1 [802,1364] : 1
x + RC_MAP_FLTMODE [810,1372] : 5
x + RC_MAP_KILL_SW [813,1375] : 10
x + RC_MAP_PITCH [821,1383] : 3
x + RC_MAP_ROLL [825,1387] : 2
x + RC_MAP_THROTTLE [827,1389] : 1
x + RC_MAP_YAW [829,1391] : 4
x + SYS_AUTOSTART [905,1541] : 24002
x + SYS_CTRL_ALLOC [913,1549] : 1
x + UAVCAN_ENABLE [934,1785] : 0

 938/1888 parameters used.

```

bkueng, 2022-02-23:
So you disabled the rate controller via `CBRK_RATE_CTRL`, which means `vehicle_torque_setpoint` isn't published. That explains it, thanks for checking. The backup scheduling will also help in that case then.