PX4/PX4-Autopilot #19195: After updating firmware to 1.12.3 from 1.11.2 on CUAV V5 plus GPS status is Disabled

vipertown, 2022-02-16:
**Describe the bug**
After updating firmware to 1.12.3 from 1.11.2 on CUAV V5 plus GPS status is Disabled.
Same thing with 1.12.2.
It seems as it is still working, but you never get "ready for flight".

Going back to 1.11.2 resolves this issue.

**To Reproduce**
1. Update firmware on CUAV V5+ to 1.12.3
2. Reboot
3. In QGC (4.2 and 4.1.2) GPS is showing up as: Disabled. 
4. 
[image]

dagar, 2022-02-16:
Can you share a full log with your parameters (https://logs.px4.io/)?

vipertown, 2022-02-18:
There is no log, PX4 v1.12.3 has not yet got flight approval of obvious reasons. 
CUAV Neo V2 GPS works, and checking GPS status in shows 3D lock and number of satelittes correctly but in QGC Sensor Status it shows up as disabled.
[image]

dagar, 2022-02-18:
> There is no log, PX4 v1.12.3 has not yet got flight approval of obvious reasons.

Understandable, but I was mainly looking for the parameters. In particular any EKF2, COM, or SYS changes.

dagar, 2022-02-18:
Is COM_ARM_WO_GPS set to 0 or 1?

vipertown, 2022-02-24:
COM_ARM_WO_GPS is set to 1. Log file attached. 

[px4_fmu-v5_multicopter_v1_12_3_LOGG.zip](https://github.com/PX4/PX4-Autopilot/files/8133220/px4_fmu-v5_multicopter_v1_12_3_LOGG.zip)

vipertown, 2022-02-24:
After setting COM_ARM_WO_GPS to 0 the GPS shows up Normal.

[image]