PX4/PX4-Autopilot #19197: Support MAVLink Standard Modes

hamishwillee, 2022-02-16:
MAVLink standard modes are described in https://github.com/mavlink/rfcs/pull/17, and are now present in development.xml

These consist of a set of standard flight modes (enum `MAV_STANDARD_MODE`) that can be queried using `MAV_CMD_REQUEST_MESSAGE` to get `AVAILABLE_MODES` (to get *all* modes), and set using `MAV_CMD_DO_SET_STANDARD_MODE`.
The current active standard mode (and custom modes) are streamed in `CURRENT_MODE`, and this should also be emitted when the mode changes.

**Describe problem solved by the proposed feature**

This service is designed to allow a ground station to interact with an unknown flight stack, because it can infer certain behaviour based on the standard mode. In this case it would mean a new flight stack or SDK could work with PX4 even if it did not know about PX4.

**Describe your preferred solution**

Implement the messages as documented, setting and emitting the correct standard mode for each PX4 custom modes.
- Before mapping the mode, sanity check that the custom modes do behave as indicated in `MAV_STANDARD_MODE`
- Note an ambiguity or questions in https://github.com/mavlink/rfcs/pull/17

**Describe possible alternatives**

Alternatives are covered in https://github.com/mavlink/rfcs/pull/17.
It is likely that in future we will also get the list of `AVAILABLE_MODES` using component information.

**Additional context**

None

hamishwillee, 2022-02-16:
FYI @julianoes - is this implementation going to be done by you?

julianoes, 2022-02-19:
@hamishwillee I might tackle the MAVSDK side.

julianoes, 2025-10-05:
@hamishwillee that's implemented in PX4, right?

closed 2025-10-05 by julianoes

hamishwillee, 2025-10-05:
Yes. Perhaps not to "total completion", but Beat did this as part of the ROS2 Modes work.