PX4/PX4-Autopilot #19198: Flight termination should not be reversible

hamishwillee, 2022-02-17:
The mavlink message [https://mavlink.io/en/messages/common.html#MAV_CMD_DO_FLIGHTTERMINATION] is supposed to terminate flight if param1 > 0.5 (see docs). 
Flight termination is then generally understood to mean "immediately and _irrevocably_ terminate the flight, bringing the vehicle to ground (i.e. not "landing" as such).

The important bit here is **irrevocably** - PX4 has added some non-spec behaviour to the message and will reverse the termination IF the `MAV_CMD_DO_FLIGHTTERMINATION` is sent with `param1 < 0.5`.

That's a bug because it is counter to general understanding of what termination means, and because it overloads the message in an inconsistent way - the theory being that values other than those specified have side effects.
I recall the original reasoning for this change was that this might allow people who assign a termination switch to undo that operation if they do it by accident. I've heard several people argue against this - termination is final. You can't undo a parachute launch :-). 
If you assign such a switch you must know the consequences. 

Anyway, the action is to:
- Undo the MAVLink incompatibility. IF `param1 < 0.5` the message should ack with MAV_RESULT_FAILED, and the vehicle should not terminate or reverse a termination.
- Set the `MAV_PROTOCOL_CAPABILITY_FLIGHT_TERMINATION` bit.

hamishwillee, 2022-02-17:
@julianoes Assigned to you :-)

ryanjAA, 2022-03-03:
I can see why this is an issue and if you had intermittent rc or something weird then after engaged it could disengage but tbh I kind of like that I can turn it on or off. I’ve hit it a couple times on test vehicle and instantly switched it back. It would have been a pain if I couldn’t and in some cases much more than a pain…. The issue is also how you control this command too. Red covered switch on your remote routed to it would be practically different than an easy to hit switch, etc. it’s a tossup but people make accidents on the practical side.

ryanjAA, 2022-03-03:
I guess the solution is to increase the time to more than 0.5 seconds (which is already possible so it’s just a personal arbitrary number) and then your logic is better. Just can’t screw up which the time amount is the fudge factor but this is important to at least think about practically speaking.

julianoes, 2022-03-06:
@ryanjAA I think you refer to the kill switch which should probably not be the same as a "termination switch". To me, the way we currently (in many cases) use the kill switch with PX4 is not a termination switch where you trigger a parachute, etc. but more of a "force disarm" switch, so a disarm swich that works in-air.

> if you had intermittent rc or something weird then after engaged it could disengage

If you have a kill or termination switch, then it must not be intermittent, in my book. Something so essential needs to be reliable, or made reliable with filtering, as you suggest.

> I’ve hit it a couple times on test vehicle and instantly switched it back.

I have seen expensive stuff fall from the sky because someone "bumped" the kill switch by accident, more than once! Therefore, in my opinion, it needs to be a big red covered switch as you suggest, or a dial instead of a switch.

My takeaway is that we could maybe leave the kill switch as is, as it can be handy, and people are used to it. However, for anything deployed, or a product, a kill switch is not something that should be easily accessible. Instead, there should be a termination switch, or other mechanism which can send the MAVLink termination command.

julianoes, 2025-10-05:
How I hate this bot.

dakejahl, 2026-10-31:
**Claude on behalf of @dakejahl**

Fixed on main. #21129 made `MAV_CMD_DO_FLIGHTTERMINATION` with param1 < 0.5 return DENIED instead of reversing termination, and #23432 stopped mode switches and disarming from leaving the termination state. #20617 sets `MAV_PROTOCOL_CAPABILITY_FLIGHT_TERMINATION`.

closed 2026-10-31 by dakejahl