PX4/PX4-Autopilot #19199: bash build_ros2_workspace.bash failed

REGATTE, 2022-02-17:
Hi,
I'm building the PX4-ROS2 Communication system using the FastDDS implementation as of the doc's in PX4 website.

Ive installed all the required libraries, dependencies and made sure the versions are correct.
Im running ROS2 Foxy and no ros1-2Bridge so ran the bash script with the following arguments:

Ubuntu 20.04
ROS2 FOXY
FastDDS 2.0.2
FastRTPS-Gen 1.0.4
Gradle 6.3

cd /to/px4_ros_com/scripts
bash build_ros2_workspace.bash --no_ros1_bridge --ros_distro foxy --ros_path ~/ros2_foxy/install/setup.bash

and i get the following errors:

```
Starting >>> px4_msgs
Finished <<< px4_msgs [25.2s]                       
Starting >>> px4_ros_com
--- stderr: px4_ros_com                              
In file included from /home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/RtpsTopics.h:39,
                 from /home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/RtpsTopics.cpp:34:
/home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_timesync.h:46:10: fatal error: timesync_Publisher.h: No such file or directory
   46 | #include "timesync_Publisher.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:536: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/RtpsTopics.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
In file included from /home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_timesync.cpp:42:
/home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_timesync.h:46:10: fatal error: timesync_Publisher.h: No such file or directory
   46 | #include "timesync_Publisher.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:523: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_timesync.cpp.o] Error 1
In file included from /home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_agent.cpp:52:
/home/ashok/px4_ros_com_ros2/build/px4_ros_com/src/micrortps_agent/microRTPS_timesync.h:46:10: fatal error: timesync_Publisher.h: No such file or directory
   46 | #include "timesync_Publisher.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/micrortps_agent.dir/build.make:497: CMakeFiles/micrortps_agent.dir/src/micrortps_agent/microRTPS_agent.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:142: CMakeFiles/micrortps_agent.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< px4_ros_com [3.09s, exited with code 2]
```

gist: `Timesync files are not being detected`

Could someone please help with fixing this.

Thanks.

REGATTE, 2022-02-18:
the error got fixed, once I set my JAVA JDK to 11. It was at 13, due to the Ubuntu.sh from PX4-Autopilot/Tools.

closed 2022-02-18 by REGATTE

FlorianPix, 2022-07-27:
I got the same Error. What exact version of the java jdk are you using ? I've got `openjdk version "11.0.15" 2022-04-19`.

zondafeng, 2023-08-02:
my java jdk is 11.0.20. but i still meet this problem