PX4/PX4-Autopilot #19203: Pool Timeout - PX4 Simulator Gazebo

Desartstudio, 2022-02-17:
Today i have start to use Gazebo for simulation of my system 

It work fine but after some minute i have a error in colsol and in gazebo and in qgroundcontrol the model are stopped 
i use OSX i9 64gb ram ..the cpu remaining to 30%...idem the gpu (ATI) the ram 40% isnt for hardware stress..

the error is :
`.......
INFO  [tone_alarm] notify positive
INFO  [commander] Armed by external command	
INFO  [navigator] Takeoff to 50.0 meters above home	
INFO  [commander] Takeoff detected	
ERROR [simulator] poll timeout 0, 22
poll timeout
ERROR [simulator] poll timeout 0, 22
poll timeout
............`
i have search and i have find this post :
[Link](https://discuss.px4.io/t/px4-poll-timeout/20484)

everyone know a solution?

danylevskyi, 2022-03-31:
Getting this issue while using Gazebo and Jmavsim.

LiuWhale, 2022-04-19:
multi vehicle sitl in ros2 has same issue.
[image]

tkhamvilai, 2022-05-17:
got the same issue on jmavsim with the newly built px4