PX4/PX4-Autopilot #19210: PX4 development environment setup error on an ARM-based Linux

mengchaoheng, 2022-02-18:
**Describe the bug**

bash ./PX4-Autopilot/Tools/setup/ubuntu.sh error

**To Reproduce**
Steps to reproduce the behavior:
1.[Download PX4 Source Code]:
git clone https://github.com/PX4/PX4-Autopilot.git --recursive

2.Run the ubuntu.sh with no arguments (in a bash shell) to install everything:
bash ./PX4-Autopilot/Tools/setup/ubuntu.sh

mengchaoheng, 2022-04-19:
I have fix it, if everyone need to install PX4 development environment ARM-based, for example，like ubuntu20.04 Virtual Machine on m1 macbook pro, that is a arm64 or aarch64 architecture cpu, before you start "bash ./PX4-Autopilot/Tools/setup/ubuntu.sh", something to be done:
1. add Ubuntu Ports to /etc/apt/sources.list
2. delete g++-multilib and gcc-multilib line in /PX4-Autopilot/Tools/setup/ubuntu.sh
3. open terminal and cd to PX4-Autopilot, and then  run "bash Tools/setup/ubuntu.sh"
4. reboot your computer
5. run "make px4_fmu-v5_default" or any target on terminal.

closed 2022-04-19 by mengchaoheng

retoc71586, 2022-06-08:
@mengchaoheng what do you mean by add Ubuntu Ports to /etc/apt/sources.list? What to modify in sources.list exactly?
Thanks for the help

mengchaoheng, 2022-06-08:
> @mengchaoheng what do you mean by add Ubuntu Ports to /etc/apt/sources.list? What to modify in sources.list exactly?
> 
> Thanks for the help
> 
> 
> 
> PS: Since I am basically following all of your issues for setting up px4 on my  ubuntu m1 parallels VM, is it there a complete guide on the process?
> 
> 

you don't have to modify sources, but sometimes it is very important. Since the arm based ubuntu is different from the x86 architecture, the sources file is different.

mengchaoheng, 2022-06-08:
> @mengchaoheng what do you mean by add Ubuntu Ports to /etc/apt/sources.list? What to modify in sources.list exactly?
> 
> Thanks for the help
> 
> 
> 
> PS: Since I am basically following all of your issues for setting up px4 on my  ubuntu m1 parallels VM, is it there a complete guide on the process?
> 
> 

https://github.com/mavlink/qgroundcontrol/issues/10205

retoc71586, 2022-06-08:
QGC I managed to install trough flatpack thanks to your other issue, however I am stil getting the error with `sudo apt-get install -y gcc-8-aarch64-linux-gnu g++-8-aarch64-linux-gnu
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package gcc-8-aarch64-linux-gnu
E: Unable to locate package g++-8-aarch64-linux-gnu
E: Couldn't find any package by regex 'g++-8-aarch64-linux-gnu'
`
Did you just delete g++-multilib and gcc-multilib line in /PX4-Autopilot/Tools/setup/ubuntu.sh to solve this?

mengchaoheng, 2022-06-08:
> QGC I managed to install trough flatpack thanks to your other issue, however I am stil getting the error with
>
>
>
> `sudo apt-get install -y gcc-8-aarch64-linux-gnu g++-8-aarch64-linux-gnu
>
> Reading [...]

Yes, you're right!

mengchaoheng, 2022-06-08:
> QGC I managed to install trough flatpack thanks to your other issue, however I am stil getting the error with
>
>
>
> `sudo apt-get install -y gcc-8-aarch64-linux-gnu g++-8-aarch64-linux-gnu
>
> Reading [...]

But, maybe you should update/upgrade?