PX4/PX4-Autopilot #19211: Add KakuteH7 BT PX4 target

abimaputra78, 2022-02-18:
For general questions please use [PX4 Discuss](http://discuss.px4.io/) or Slack (you can find an invite link on this project README).

**Describe problem solved by the proposed feature**
A multi fuctional multicopter to use full acro and autopilot feature from PX4

**Describe your preferred solution**
An Acro means full acro capabilities without accelerometer stabilizer function. Like in Betaflight and other race firmware and airmode

**Describe possible alternatives**
A reliable return to home and reliable waypoint. 

**Additional context**

bkueng, 2022-02-18:
KakuteH7 is already supported, and all of your described features as well. What are you missing in particular?

abimaputra78, 2022-02-18:
Thanks.
How can I download that Sir? I haven't found in Git nor px4 web

bkueng, 2022-02-21:
You have to use master for now as there's not yet any release.

closed 2025-10-08 by hamishwillee