PX4/PX4-Autopilot #19230: mavros/local_position/odom and pose giving random wrong values

bhaskar-glitch, 2022-02-20:
Hi, I'm facing a problem while using odometry for the position of the drone..I'm using realsense d435i, jetson xavier nx, and pixhawk 2.4.8 with px4 firmware....I'm using a bridge VIO to provide odometry data to px4..how the issue I'm facing right now is that bridge is sending data to /mavros/odometry/out and then odometry out is sending those data to mavros and further those data get into /mavros/local_position/odom and pose...but the position data (X,y,z) from the VINS(odometry)is correct till /mavros/odometry/out(topic) for after that when I'm doing rostopic echo of /mavros/local_positon/odom and pose it is giving some random wrong values...I don't know why because at /odometry/out topic values are very fine but after going to mavros node it gets wrong!! also, I changed all the parameters which is required for vision pose estimate like(EKF2_AID_MASK=24, EKF2_HGT_MODE=VISION and turned off the barometer)!!
Please help regarding the issue that why I'm getting that random wrong values.

ROSGRAPH
======
[image]

bresch, 2022-02-21:
Do you maybe have a log file from the flight controller (`.ulg`)?

bhaskar-glitch, 2022-02-21:
> Do you maybe have a log file from the flight controller (`.ulg`)?

Actually we haven't taken any flight..How can i get log file??...I have just checked the data in the topics by ros...mavros/local_position/odom and pose are wrong and behaving abnormally...but the topic mavros/odometry/out have right values..I don't know why the values are becoming wrong after going to mavros node...

farhangnaderi, 2022-03-08:
@bhaskar-glitch you can upload your log files here:
https://review.px4.io/

This is also the link to acquire the logs:
https://docs.px4.io/master/en/getting_started/flight_reporting.html

bhaskar-glitch, 2022-03-13:
> @bhaskar-glitch you can upload your log files here: https://review.px4.io/
> 
> This is also the link to acquire the logs: https://docs.px4.io/master/en/getting_started/flight_reporting.html

i have uploaded the log file on the link also 
https://drive.google.com/file/d/14Jhdzz517YFApU6XJiRoiILGIwc3TOWz/view?usp=sharing  
this is the log file link