PX4/PX4-Autopilot #19231: HolyBro Pixhawk4 mini has serial output freeze through micro-b usb port with firmware v1.12.3

comperem, 2022-02-20:
**Bug Descrition**
Serial output via micro-usb to QGroundControl (or pymavlink) freezes intermittently for N seconds during motion with the 1.12.3 firmware. This did not occur with the v1.11.3 firmware.

**To Reproduce**
Steps to reproduce the behavior:
1. connect PX4 mini to PC, on a desk, separate from any vehicle, and open QGroundControl
2. While watching either GLOBAL_POSITION_INT or LOCAL_POSITION_INT in MAVLink Inspector, rotating the device causes intermittent update freezes. The serial output is paused, or delayed or frozen for 1/2 to 10 to 20 seconds.
3. Uploaded an example mission but the serial output freeze is not reflected in the mission log.

**Expected behavior**
Sitting on the desk, watching signals from a Pixhawk4mini should not cause freezing or frozen serial updates in certain signals. The delay can range from 0.5seconds to 20 seconds. It's unclear why other than it's association with moderate, smooth motion while moving the device on the desk, in orientation mainly.

**Log Files and Screenshots**
Here's the position record from the usb serial port as read by pymavlink collecting LOCAL_POSITION_NED. The missing data are the freezing sections that correspond to smooth turning motion. Wiring is all good and strain relief on the USB connector does not help.

LOCAL_POSITION_NED plot from serial:
[image]

Time history from 5 separate time records on the Pixhawk and Laptop:
[image]

Here's the flight log from the same drive. Unfortunately it looks beautiful. There is no hint of any problems in the Pixhawk flight log:
https://review.px4.io/plot_app?log=48ed7bff-d92d-4c02-988e-84ccc4d5fec2

The problem is entirely in the usb serial output port.

Unusual scenario: This device is being used for data collection. It is not being armed and is not controlling any actuators.

comperem, 2022-02-20:
by the way, this wonky behavior where usb serial port output hangs in firmware [v1.12.1](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.1) but not [v1.11.3](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.11.3) or [v1.12.0](https://github.com/PX4/PX4-Autopilot/releases/tag/v1.12.0).

Firmware filename: `px4_fmu-v5_default.px4`

The freezing, or hanging serial behavior can be seen very easily, right on your desktop and near a window for GPS. Opening QGroundControl and watching MAVLink Inspector, then moving the device mainly in orientation, the delays, or serial hangs can be seen quite clearly.

Also, the message I'm most interested in, which is GLOBAL_POSITION_INT, shows up with v1.12.1 and onward.

dagar, 2022-02-20:
Out of curiosity have you tried current master (same as flashing development through QGC)? It's going to become v1.13 in the near future.

dagar, 2022-02-20:
The "freezing" is when the estimator position and velocity goes invalid.
[image]

One thing that comes to mind is that the system is really built around the typical usage scenario of arming to operate. When disarmed the vehicle never goes to the "in_air" state that the EKF checks in a few different places.

What's your full use case? If the system is never going to be properly armed we should talk about land detector changes (estimator in_air flag).

dagar, 2022-02-20:
More specifically I'm talking about these additional checks on the validity flags that are only active when disarmed (!in_air). 

https://github.com/PX4/PX4-Autopilot/blob/cc6c6c3b8cb56f79eb6bb09d8d4bc7535a30b184/src/modules/ekf2/EKF2.cpp#L883-L886

https://github.com/PX4/PX4-Autopilot/blob/cc6c6c3b8cb56f79eb6bb09d8d4bc7535a30b184/src/modules/ekf2/EKF2.cpp#L731

Then in the mavlink streams those flags are checked again. https://github.com/PX4/PX4-Autopilot/blob/cc6c6c3b8cb56f79eb6bb09d8d4bc7535a30b184/src/modules/mavlink/streams/LOCAL_POSITION_NED.hpp#L65

comperem, 2022-02-21:
> Out of curiosity have you tried current master (same as flashing development through QGC)? It's going to become v1.13 in the near future.

Hello, mm, maybe yes, I tried the v1.12.3 which seems to be the latest stable. I did not try any betas or custom builds from source. Just the standard stable applied by QGroundControl.

comperem, 2022-02-21:
> The "freezing" is when the estimator position and velocity goes invalid, there are additional preflight checks applied to the position/velocity validity when disarmed/landed.
> 
> [image]
> 
> One thing that comes to mind is that the system is really built around the typical usage scenario of arming to operate. When disarmed the vehicle never goes to the "in_air" state that the EKF checks in a few different places.
> 
> What's your full use case? If the system is never going to be properly armed we should talk about land detector changes (estimator in_air flag).

Ok, ok, this is helpful. I've been wondering if I should attempt to arm the system via the python data acquisition program. Those checks you found in the next message were kind of what I was wondering about. Arming is the standard use case for a flight controller and that recently made me realize I could be missing some checks, just as you pointed out.

comperem, 2022-02-21:
> More specifically I'm talking about these additional checks on the validity flags that are only active when disarmed (!in_air).
> 
> https://github.com/PX4/PX4-Autopilot/blob/cc6c6c3b8cb56f79eb6bb09d8d4bc7535a30b184/src/modules/ekf2/EKF2.cpp#L883-L886
> 
> https://github.com/PX4/PX4-Autopilot/blob/cc6c6c3b8cb56f79eb6bb09d8d4bc7535a30b184/src/modules/ekf2/EKF2.cpp#L731
> 
> Then in the mavlink streams those flags are checked again potentially blocking publication.
> 
> https://github.com/PX4/PX4-Autopilot/blob/cc6c6c3b8cb56f79eb6bb09d8d4bc7535a30b184/src/modules/mavlink/streams/LOCAL_POSITION_NED.hpp#L65

These checks and also the checks required to calibrate sensors before arming may, indeed, eliminate the serial message freezes.

It's unclear but I'll look into how difficult or feasible it is to provide a software, programmatic sequence for a data acquisition application. 

This device was intended to provide vehicle velocity in the NED frame. We've been using the LOCAL_POSITION_NED message but recently discovered those velocities are terribly flawed. That [bug report is here](https://discuss.px4.io/t/holybro-pixhawk-4-mini-ned-velocity-bug-report-in-vx-and-vy/26160) and has nothing to do with serial message blockage. The 4th image down (slide 6 of 9) shows the failing velocities.

2 serious failures in a row for a mature device is not expected.

dagar, 2022-02-21:
> This device was intended to provide vehicle velocity in the NED frame. We've been using the LOCAL_POSITION_NED message but recently discovered those velocities are terribly flawed. That [bug report [...]

I would propose we get the validity checks fixed for this use case and then re-evaluate. I could add something quickly to explicitly support this use case.

Then if you're able to repeat the test we can log (.ulg) all raw sensor data at full rate so that if any issues remain we can replay the data offline and debug/identify/fix the root cause via EKF replay.

dagar, 2022-02-21:
> 2 serious failures in a row for a mature device is not expected.

I agree, that kind of thing would definitely not go unnoticed in real flight logs.

dagar, 2022-02-21:
@comperem I've made a small change in https://github.com/PX4/PX4-Autopilot/pull/19239 that should improve this particular logging case if you don't set a vehicle (SYS_AUTOSTART 0). The system won't be running the land detector and the EKF will now assume you're possibly in air. Init will be slightly slower (10s), but you probably won't even notice.

comperem, 2022-02-21:
Hello @dagar , thank you. I'm just now trying out the calibration and arming procedures programmatically. Our use-case needs to start at power-up and stream data.

for this change you've made, I'll have to look. we have been using a quadcopter for a nominal setting but it's not important, really, because it's not controlling anything. I'll look for SYS_AUTOSTART.

comperem, 2022-02-21:
with SYS_AUTOSTART set to 0, QGroundControl says Airframe is not setup and nothing else can be adjusted until this is setup.

dagar, 2022-02-21:
> with SYS_AUTOSTART set to 0, QGroundControl says Airframe is not setup and nothing else can be adjusted until this is setup.

That's only a QGC display thing. Short term you could work around this by having the sensors already calibrated on your quad, then changing the parameter SYS_AUTOSTART 0 manually (don't click the airframe box in QGC).

If you use the change in https://github.com/PX4/PX4-Autopilot/pull/19239 with no airframe set (SYS_AUTOSTART 0) you can ignore things like flight modes, battery power, etc.

comperem, 2022-02-21:
Ok, understood. This is helpful. It's not entirely clear what QGroundControl does and what the pymavlink library does.

comperem, 2022-02-23:
@dagar , thanks for getting that new commit put together and approved.

what's the general timeframe for the next release? Are the auto-built pre-release versions? It's unclear where to find the firmware binary for `px4_fmu-v5_default.px4`.

should I wait for the full release to test?

comperem, 2022-03-08:
dead thread.

No more information available and no way to follow up.

???

dagar, 2022-03-08:
Yes there are nightly builds, it's what you get when you flash "development" through QGC.

Here's the link if you want to grab it directly. https://px4-travis.s3.amazonaws.com/Firmware/master/px4_fmu-v5_default.px4
It corresponds to the latest build from the master branch.

comperem, 2022-03-09:
ahhhh.. I had no idea. I searched and didn't find the right place to look.

Thank you. We will try this immediately.

comperem, 2022-03-24:
We tested the developmental version of Pixhawk firmware on a HolyBro Pixhawk 4 mini on 10 March 2022. the serial port did not freeze (which is good) and was also not armed.

This use-case of data collection and motion measurement without arming the pixhawk seems to be working now.

There is another issue related to drifting yaw angle while at zero velocity. This is a new issue I'll open now.

This case can be closed.

Thanks for your help

closed 2022-03-24 by comperem