PX4/PX4-Autopilot #19266: make px4_sitl gazebo in m1 pro macbookpro error

mengchaoheng, 2022-02-27:
**Describe the bug**

/usr/local/include/gazebo-11/gazebo/common/ffmpeg_inc.h:32:10: fatal error: 'libavcodec/avcodec.h' file not found
#include <libavcodec/avcodec.h>
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build 

**To Reproduce**
Steps to reproduce the behavior:
1. A new computer follow:  https://docs.px4.io/master/en/dev_setup/dev_env_mac.html

**Expected behavior**
A clear and concise description of what you expected to happen.

**Log Files and Screenshots**
*Always* provide a link to the flight log file:
- Download the flight log file from the vehicle ([tutorial](https://docs.px4.io/master/en/getting_started/flight_reporting.html)).
- Share the link to a log showing the problem on [PX4 Flight Review](http://logs.px4.io/).

Add screenshots to help explain your problem.

mengchaoheng@mengchaohengdeMacBook-Pro PX4-Autopilot % make px4_sitl gazebo                   
-- PX4 version: v1.13.0-alpha1-4371-gc2c455be0d
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.9.10", minimum required is "3") 
-- PX4 config file: /Users/mengchaoheng/Downloads/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is AppleClang 13.0.0.13000029
-- The C compiler identification is AppleClang 13.0.0.13000029
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- ROMFS: ROMFS/px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default
[0/853] git submodule src/drivers/gps/devices
[12/853] git submodule Tools/sitl_gazebo
[15/853] git submodule src/modules/mavlink/mavlink
[19/853] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /Users/mengchaoheng/Downloads/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 228 MAVLink message types in 4 XML files
[... 222 lines cut ...]
1 error generated.
[59/130] Building CXX object CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o
FAILED: CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o 
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/c++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_IOSTREAMS_DYN_LINK -DBOOST_IOSTREAMS_NO_LIB -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_TEST_DYN_LINK -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DLIBBULLET_VERSION=3.20 -DLIBBULLET_VERSION_GT_282 -Dgazebo_camera_manager_plugin_EXPORTS -I/Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_gazebo/include -I/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/build_gazebo -I/usr/local/include/eigen3/eigen3 -I/usr/local/include/gazebo-11/gazebo/msgs -I/Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/mavlink -I/Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_gazebo/external/OpticalFlow/include -I/usr/local/Cellar/libffi/3.4.2/include -I/usr/local/Cellar/gstreamer/1.18.5/include/gstreamer-1.0 -I/usr/local/Cellar/glib/2.70.4/include -I/usr/local/Cellar/glib/2.70.4/include/glib-2.0 -I/usr/local/Cellar/glib/2.70.4/lib/glib-2.0/include -I/usr/local/opt/gettext/include -I/usr/local/Cellar/pcre/8.45/include -I/usr/local/Cellar/gst-plugins-base/1.18.5/include/gstreamer-1.0 -isystem /usr/local/include -isystem /usr/local/include/gazebo-11 -isystem /usr/local/Cellar/bullet/3.21/include/bullet -isystem /usr/local/Cellar/bullet/3.21/include -isystem /usr/local/include/simbody -isystem /usr/local/Cellar/sdformat9/9.7.0/include/sdformat-9.7 -isystem /usr/local/include/ignition/math6 -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_10/include/OGRE/Paging -isystem /usr/local/include/ignition/transport8 -isystem /usr/local/include/ignition/msgs5 -isystem /usr/local/include/ignition/common3 -isystem /usr/local/include/ignition/fuel_tools4 -iframework /usr/local/opt/qt@5/lib -isystem /usr/local/opt/qt@5/lib/QtCore.framework/Headers -isystem /usr/local/opt/qt@5/./mkspecs/macx-clang -isystem /usr/local/include/eigen3 -isystem /usr/local/Cellar/opencv/4.5.4_4/include/opencv4 -isystem /usr/local/Cellar/fcl/0.7.0/include -isystem /usr/local/Cellar/assimp/5.2.2/include -isystem /usr/local/include/sdformat-9.7/sdf/.. -isystem /usr/local/include/ignition/cmake2 -O2 -g -DNDEBUG -arch x86_64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX12.1.sdk -fPIC -F/Library/Developer/CommandLineTools/SDKs/MacOSX11.sdk/System/Library/Frameworks  -Wno-deprecated-declarations -Wno-address-of-packed-member -std=gnu++17 -MD -MT CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o -MF CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o.d -o CMakeFiles/gazebo_camera_manager_plugin.dir/src/gazebo_camera_manager_plugin.cpp.o -c /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_camera_manager_plugin.cpp
In file included from /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_gazebo/src/gazebo_camera_manager_plugin.cpp:18:
In file included from /Users/mengchaoheng/Downloads/PX4-Autopilot/Tools/sitl_gazebo/include/gazebo_camera_manager_plugin.h:24:
In file included from /usr/local/include/gazebo-11/gazebo/gazebo.hh:20:
In file included from /usr/local/include/gazebo-11/gazebo/gazebo_core.hh:19:
In file included from /usr/local/include/gazebo-11/gazebo/common/common.hh:53:
/usr/local/include/gazebo-11/gazebo/common/ffmpeg_inc.h:32:10: fatal error: 'libavcodec/avcodec.h' file not found
#include <libavcodec/avcodec.h>
         ^~~~~~~~~~~~~~~~~~~~~~
1 error generated.
ninja: build stopped: subcommand failed.
FAILED: external/Stamp/sitl_gazebo/sitl_gazebo-build /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/external/Stamp/sitl_gazebo/sitl_gazebo-build 
cd /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/build_gazebo && /usr/local/Cellar/cmake/3.22.2/bin/cmake --build /Users/mengchaoheng/Downloads/PX4-Autopilot/build/px4_sitl_default/build_gazebo -- -j 2
ninja: build stopped: subcommand failed.
make: *** [px4_sitl] Error 1

**Drone (please complete the following information):**
- Describe the type of drone.
- Photo of the IMU / autopilot setup if possible.

**Additional context**
Add any other context about the problem here.

mengchaoheng, 2022-02-28:
[image]

mengchaoheng, 2022-02-28:
@dagar help me please!

ThomasDebrunner, 2022-03-01:
Are you running this in rosetta emulation or not? Afaik, gazebo does not work yet on arm macs

mengchaoheng, 2022-03-01:
> Are you running this in rosetta emulation or not? Afaik, gazebo does not work yet on arm macs

But still fail when build gazebo simulation

mengchaoheng, 2022-03-01:
> Are you running this in rosetta emulation or not? Afaik, gazebo does not work yet on arm macs

It looks like the problem is with the gazebo version and the plugin.

mengchaoheng, 2022-03-03:
it work now! @ThomasDebrunner @dagar 

https://github.com/PX4/PX4-Autopilot/issues/18413

mengchaoheng, 2022-03-03:
by the step in https://github.com/PX4/PX4-Autopilot/issues/18413

[image]

ThomasDebrunner, 2022-03-03:
I'm glad!

closed 2022-03-03 by ThomasDebrunner

Hexiang-W, 2022-11-25:
@mengchaoheng Hello mate, I wonder how you solve "make px4_sitl gazebo" error. It seems like i cannot even make px4_sitl_default.

`hexiang_w@dhcp-10-248-215-76 PX4-Autopilot % make px4_sitl_default                  
-- PX4 version: v1.13.0-beta1-1239-g85e9c17a5e
-- Found PythonInterp: /usr/local/bin/python3 (found suitable version "3.10.8", minimum required is "3") 
-- PX4 config file: /Users/hexiang_w/PX4-Autopilot/boards/px4/sitl/default.px4board
-- PLATFORM posix
-- ROMFSROOT px4fmu_common
-- TESTING y
-- ETHERNET y
-- PX4 config: px4_sitl_default
-- PX4 platform: posix
-- PX4 lockstep: enabled
-- cmake build type: RelWithDebInfo
-- The CXX compiler identification is AppleClang 14.0.0.14000029
-- The C compiler identification is AppleClang 14.0.0.14000029
-- The ASM compiler identification is Clang with GNU-like command-line
-- Found assembler: /Library/Developer/CommandLineTools/usr/bin/cc
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Building for code coverage
-- ccache enabled (export CCACHE_DISABLE=1 to disable)
-- Looking for ignition-transport8 -- found version 8.3.0
-- Searching for dependencies of ignition-transport8
-- Found Protobuf: /usr/local/lib/libprotobuf.dylib (found suitable version "3.21.9", minimum required is "3") 
-- Config-file not installed for ZeroMQ -- checking for pkg-config
-- Checking for module 'libzmq >= 4'
--   Found libzmq , version 4.3.4
-- Found ZeroMQ: TRUE (Required is at least version "4") 
-- Found UUID: TRUE  
-- Looking for ignition-msgs5 -- found version 5.11.0
-- Searching for dependencies of ignition-msgs5
-- Looking for ignition-math6 -- found version 6.13.0
-- Searching for dependencies of ignition-math6
-- Checking for module 'tinyxml2'
--   Found tinyxml2, version 9.0.0
-- ROMFS: ROMFS/px4fmu_common
-- Configuring done
-- Generating done
-- Build files have been written to: /Users/hexiang_w/PX4-Autopilot/build/px4_sitl_default
[0/2] Re-checking globbed directories...
[0/880] git submodule src/modules/microdds_client/Micro-XRCE-DDS-Client
[9/880] git submodule src/drivers/gps/devices
[11/880] git submodule src/modules/mavlink/mavlink
[13/880] Generating Mavlink development: src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Parsing /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/development.xml
Validating /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Parsing /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/standard.xml
Validating /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Parsing /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/common.xml
Validating /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Parsing /Users/hexiang_w/PX4-Autopilot/src/modules/mavlink/mavlink/message_definitions/v1.0/minimal.xml
Merged enum MAV_CMD
Merged enum MAV_CMD
Found 232 MAVLink message types in 4 XML files
Generating C implementation in directory /Users/hexiang_w/PX4-Autopilot/build/px4_sitl_default/mavlink/development
Generating C implementation in directory /Users/hexiang_w/PX4-Autopilot/build/px4_sitl_default/mavlink/standard
Generating C implementation in directory /Users/hexiang_w/PX4-Autopilot/build/px4_sitl_default/mavlink/common
Generating C implementation in directory /Users/hexiang_w/PX4-Autopilot/build/px4_sitl_default/mavlink/minimal
Copying fixed headers for protocol 2.0 to /Users/hexiang_w/PX4-Autopilot/build/px4_sitl_default/mavlink
[32/880] Generating Mavlink uAvionix: src/modules/mavlink/mavlink/message_definitions/v1.0/uAvionix.xml
[... 84 lines cut ...]
                        pos += sprintf(buf + pos, "%s", __px4_log_level_color[level]);
                               ^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 error generated.
[339/880] Building CXX object platforms/common/uORB/uORB_tests/CMakeFiles/modules__uORB__uORB_tests.dir/uORBTest_UnitTest.cpp.o
FAILED: platforms/common/uORB/uORB_tests/CMakeFiles/modules__uORB__uORB_tests.dir/uORBTest_UnitTest.cpp.o 
/usr/local/bin/ccache /Library/Developer/CommandLineTools/usr/bin/c++ -DCONFIG_ARCH_BOARD_PX4_SITL -DENABLE_LOCKSTEP_SCHEDULER -DMODULE_NAME=\"uorb_tests\" -DPX4_MAIN=uorb_tests_app_main -D__CUSTOM_FILE_IO__ -D__PX4_DARWIN -D__PX4_ [... cut ...]
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")
^
/Library/Developer/CommandLineTools/SDKs/MacOSX13.0.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
1 error generated.
[344/880] Generating uORB topic ucdr headers
ninja: build stopped: subcommand failed.
make: *** [px4_sitl_default] Error 1`