PX4/PX4-Autopilot #19268: Sending MAV_CMD_NAV_TAKEOFF while flying in other flight modes pushes the drone towards vec = (1, 1, 0)

squizz617, 2022-02-28:
**Describe the bug**
When the drone is flying in MANUAL / ALTCTL / POSCTL / ACRO (and probably more?) mode and receives a MAV_CMD_NAV_TAKEOFF message (to set mode to TAKEOFF), it flies towards (x, y, z) = (1, 1, 0) direction.

**To Reproduce**
Steps to reproduce the behavior:
[terminal 1] make px4_sitl_rtps gazebo -j4
[terminal 2] Download and run the [PoC](https://gist.github.com/squizz617/f1cb009d7e4ff730492a6b9243bedf6a)

**Expected behavior**
As described in the [documentation](https://docs.px4.io/master/en/flight_modes/takeoff.html), the drone should get to the target altitude and hold. No movement in the X and Y direction is expected.

**Log Files and Screenshots**
- Flight review: https://logs.px4.io/plot_app?log=a0ec7e87-0bb5-44bd-8397-5af27a1d9882

**Drone (please complete the following information):**
- Build: `px4_sitl_rtps`
- tested commit: `a2645418`

xiaobai-cai-hold, 2022-03-01:
hello , do you use  ros2 to send MAV_CMD_NAV_TAKEOFF message? 
I used ros2 to send MAV_CMD_NAV_TAKEOFF message to px4 , it can be set to takeoff mode, but it can't takeoff.
I found the documentation you described above, takeoff need some params, however I did't find any ways to set takeoff params?

Can you share your ways that control the vehicle to takeoff ? thanks in advance

squizz617, 2022-03-01:
Hi @xiaobai-cai-hold.
As shown in the PoC above, I used MAVLink (through pymavlink APIs), not ros2, for setting takeoff mode.
I thought with ROS2 and rtps bridge, we can only use the offboard mode.. anyways, it seems that you should create a separate issue for your question.

mcsauder, 2022-03-02:
I think I've replicated this issue. https://review.px4.io/plot_app?log=aa495108-dcc2-4b03-bf98-a87ab54af336
Glancing at the takeoff logic I can't see where the takeoff setpoint is being generated, however, I also can't determine what the difference from 1.12.3 is.