PX4/PX4-Autopilot #19270: Gazebo simulation bug when landing on a box object

Nyaruko, 2022-03-01:
**Describe the bug**
During the gazebo (SITL) simulation, if we would like to land a drone on a box, it causes the accelerometer to get noisey readings and might even cause the EKF to diverge.

**To Reproduce**
Steps to reproduce the behavior:
1. Start the Gazebo simulator make px4_sitl_default gazebo
2. Start the GCS
3. Takeoff the drone
4. Drag a box from the gazebo selection tab and put it under the drone.
5. Land the drone on that box.

**Expected behavior**
When the box and the drone have surface contacts, the simulator produces very large noisy readings for the simulated accelerometer, the noise level is so high that it might cause the EKF to diverge.

Jaeyoung-Lim, 2022-03-01:
@Nyaruko I dont think this is PX4 related, but more coming from the contact model of gazebo.

Which dynamics engine are you using gazebo? It might help to switch to something like DART

Nyaruko, 2022-03-01:
Thanks! But I have tried the DART but the problem remains. Any physics parameter tuning suggestions are welcome.

Jaeyoung-Lim, 2022-03-01:
@Nyaruko You can also fix the box if you don't need the box moving. I think that would help on making the contact problem easier

Jaeyoung-Lim, 2022-03-01:
@Nyaruko I think you can still do this, you just need to be careful on what you are landing on.

Nyaruko, 2022-03-01:
I am sorry, but could you kindly elaborate a bit more about "what you are landing on" ? Do I need to make the model into a specific shape or tune some specific parameters?

Jaeyoung-Lim, 2022-03-26:
@Nyaruko Yes, and this is more coming from the limitation of gazebo itself (or using ODE with gazebo)

closed 2022-04-01 by Jaeyoung-Lim