PX4/PX4-Autopilot #19277: Motors Spin Up to 1500 PWM Once Armed (Control Allocation stable V1.12.3)

ErcBunny, 2022-03-03:
Hi, I am using stable release v1.12.3. It seems that this issue (https://github.com/PX4/PX4-Autopilot/issues/17220) is still not fixed. I can get the right `actuator_controls_4` but `actuator_outputs` still step up to 1512 when armed, and its data do not respond to changes of `actuator_controls_4`. Airframe: `6003_hexa_x_ctrlalloc`.

BTW, I notice the line `mc_rate_control stop` in `rc.ctrlalloc`, but `mc_rate_control status` returns `running`. This behavior is observed in SITL. I am not sure if this occurs on real hardware too.

_Originally posted by @ErcBunny in https://github.com/PX4/PX4-Autopilot/issues/17220#issuecomment-1057705942_

ErcBunny, 2022-03-03:
> BTW, I notice the line `mc_rate_control stop` in `rc.ctrlalloc`, but `mc_rate_control status` returns `running`. This behavior is observed in SITL. I am not sure if this occurs on real hardware too.

Also if `mc_rate_control` really `stops` (by entering this command in pxh terminal), then the vehicle may have trouble taking off...

bkueng, 2022-03-03:
Please try with master, there has been a lot of development in that area.

ErcBunny, 2022-03-03:
Hi, @bkueng . Unfortunately, it is quite difficult for me to switch to the master branch for now, since I've made a lot of changes on stable v1.12.3, which may cause tons of conflicts.

Here (https://github.com/PX4/PX4-Autopilot/issues/17220#issuecomment-817293197), @jlecoeur commented that there is a possible workaround by connecting ESCs to the FMU side (which is the aux hardware port I guess). I haven't dug too deep into the mixer system yet. What is the "appropriate mixer on FMU"? How to set it up? Any examples?

Thanks!

ryanjAA, 2022-03-03:
If you’re using control allocation there is no mixer and this does work correctly on master (been testing it all week) but As suggested you need to use master for this. If you put esc (one or many) on aux that is fine (I do that) just set pwm_aux_dis to something other than 1500 which does spin them initially and you’re then fine. I think it’s important to note that this is a new feature and still being further developed so if something is missing or not working how youd like, updating to master or pulling in all the changes are the only options but it does work with the current iterations of it.

ryanjAA, 2022-03-03:
I am not sure if you can use ctrl allocation and a mixer on aux (or main mixer with main outputs disabled) at the same time. That isn’t the idea or spirit of it I believe (but I am not 100% on that comment) and if you could, I don’t know if you still can but if you have the outputs disabled from the ui then maybe it is possible. You could manually load an aux mixer while control alloc is setup but all aux (or main) outputs are disabled from the ui and see. Write one (or copy an existing one and change it to aux). Then load it automatically or via console. https://docs.px4.io/master/en/concept/mixing.html

ErcBunny, 2022-03-05:
Hi @ryanjAA , thanks for sharing info about the development of the master branch. I am able to get v1.12.3 working with the "direct mixer" by using `actuator_controls_1` instead of `actuator_controls_4` while changing the "control group index" in `AllocatedActuatorMixer.cpp` from `4` to `1`. (https://github.com/ErcBunny/PX4-Autopilot/commit/2ca5e0fc9f4a6c596907d50b5af58a71e95d9717)

Now I can get almost the right PWM outputs on main outputs (hooked to the IO board), except that pin 4 ramps up to 1500 and then falls back to the expected value. Any idea why this is happening?

[image]

ErcBunny, 2022-03-07:
> Now I can get almost the right PWM outputs on main outputs (hooked to the IO board), except that pin 4 ramps up to 1500 and then falls back to the expected value. This behaviour is not observed in sitl. Any idea why this is happening?

Maybe it is because `actuator_controls_0` has higher priority than `actuator_controls_1`. Solved by using `actuator_controls_0` with control allocator and direct mixer.

Closing this issue.

closed 2022-03-07 by ErcBunny